UZ-SLAMLab/ORB_SLAM3 #233: question about Ros examples about kinect2 camera

xiaoyuamw, 2021-01-08:
Hi,@jjgr3496

When i compling and running  RGB_D Node of the Ros Examples  of ORB-SLAM3, using RGB-D camera named kinect2 , there is something wrong with me! 
I have complied the project successfuly , then  changed the settings file and RGB-D topics with my own camera according to the "README.md" file, however when i run the sample with the order "rosrun ORB_SLAM3 RGBD /home/xy/catkin_SLAM3/src/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/xy/catkin_SLAM3/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/kinect1.yaml" the error occured! 
The output information as fellows:
-------------------------------------------------------------------------------------------------------
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
*Not Supported Camera Sensor*
Check an example configuration file with the desired sensor
- fps: 30
- color order: RGB (ignored if grayscale)
Segmentation fault (core dumped)
----------------------------------------------------------------------------------------------------
I don't know why such error occurs, and it's amazing that the same operation with ORB_SLAM2 can work successfully!
Can you help me to solve this error? Thank you very much!

Best Regards,
Xiaoyuamw

AnhTuDo1998, 2021-01-11:
Hi do you mind sharing your kinect1.yaml file ?

AnhTuDo1998, 2021-01-11:
Excerpt from src/Tracking.cc, inside function ORB_SLAM3::Tracking::ParseCamParamFile(cv::FileStorage &)

`else
    {
        std::cerr << "*Not Supported Camera Sensor*" << std::endl;
        std::cerr << "Check an example configuration file with the desired sensor" << std::endl;
    }`

Looks like your Camera.type parameter in the above yaml file does not match with the type defined by the author. Maybe you can consider adding your own implementation of your camera.type based on the existing two ('PinHole' and 'KannalaBrandt8'). Should be straightforward, hopefully

xiaoyuamw, 2021-01-11:
Hi, @AnhTuDo1998 

Thank you very much for your reply!
I have changed the relevant parameter according to the idea you provided, and it worked!
Thanks again!

Best Regards,
Xiaoyuamw

closed 2021-01-11 by xiaoyuamw

AnhTuDo1998, 2021-01-11:
Glad that it help :)

xiaoyuamw, 2021-01-11:
Hi，do you know why there are very few key points or nothing when detecting a person, meanwhile many key points can acquire when detecting the objects with the same distance?

AnhTuDo1998, 2021-01-11:
Hi, I assume you are using SLAM with your own camera? Would it be possible to have a  screenshot of keyframes on human and keyframes on object? Are there a lot of edges and corners on the object scenes as comparing to the human's ?

xiaoyuamw, 2021-01-11:
> Hi, i are using SLAM with my own camera. The screenshot and the kinect1.yaml file are both provide to you, maybe you could check them for me and provide some ideas to optimize the project.
[kinect1.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/5795951/kinect1.txt)

[image]

AnhTuDo1998, 2021-01-12:
You can try wearing other type of clothes, maybe the black shirt is bad for tracking. 

However, i don't see this as a bad thing as people are considered as dynamic object and should not be tracked in your pose map anyway (moving object can make the camera pose move and clutter the map with  keyframes we dont want to track). 

I am not very sure if Visual Inertial Odometry  (used in this project) actually helped resolved this problem and that is why you are not tracked in the scene or not.

xiaoyuamw, 2021-01-12:
Ok, thank you very much for your help, i  will make further study about it. 

Best Regards