UZ-SLAMLab/ORB_SLAM3 #236: Connection ORB and Airsim Camera with Ros

yamakb, 2021-01-12:
Hi, 

I was trying to connect ORB's viewer and airsim with ros. 

Airsim's publishers are: 
            

-              /airsim_node/drone_1/front_left_camera/Scene

-              /airsim_node/drone_1/front_right_camera/Scene
-              /airsim_node/drone_1/imu/Imu

ORB's Subscribers (listeners) are : 
            

-              /camera/left/image_raw
-              /camera/right/image_raw
-              /imu

So, I want to make a connection between those topics to see airsim vehicle's camera on ORB's viewer. Until I create my settings (.yaml file), I was trying to use EuRoC.yaml. 

I have tried to run : 
                  `rosrun ORB_SLAM3 Stereo_Inertial ORB_SLAM3/Vocabulary/ORBvoc.txt ORB_SLAM3/Examples/Stereo-Inertial/EuRoC.yaml /airsim_node/drone_1/front_left_custom/Scene:=/camera/left/image_raw true
`

also tried to run : 
                  `roslaunch airsim_ros_pkgs airsim_node.launch /airsim_node/drone_1/front_left_custom/Scene:=/camera/left/image_raw
`

However when I checked whether they matched, their publishers and subscribers are not matched. 
When I applied code for EuRoC bag, it works. But I need to receive data from airsim drone camera to work on ORB, not EuRoC's camera.

How can I receive data from Airsim drone camera publishers to ORB camera subscribers ? 
Thank you.

AnhTuDo1998, 2021-01-12:
Not sure if this is applicable for ORB-SLAM3 as i did this on ORB-SLAM2 but you can try adding this lines inside launch file, inside the <node> </node> tags as follow:

`<node ...>
<remap from="camera/rgb/image_raw" to="camera/color/image_raw" />
<remap from="camera/depth_registered/image_raw" to="camera/aligned_depth_to_color/image_raw"
</node>`

Explanation on the remapping order is as shown on [this](http://wiki.ros.org/roslaunch/XML/remap) ROS tutorial

yamakb, 2021-01-12:
> <remap from="camera/rgb/image_raw" to="camera/color/image_raw" />
> <remap from="camera/depth_registered/image_raw" to="camera/aligned_depth_to_color/image_raw"

Thanks for reply. The link suggests that we should add this code to subscriber's launch file (ORB in our case). I could not reach any ORB launch file directly, so I added this code to airsim_node.launch. After checking rostopic info of ORB cameras, it seems that publishers and subscribers have matched. However, now, airsim camera nodes are gone. I cannot see them in rostopic list. 

So, if it is possible, how can I change ORB launch file for adding this code? Because there is not any .launch file in ORB directories. If I should apply this method airsim_node.launch, why airsim camera topics are gone?
Thank you again.

yamakb, 2021-01-13:
I cannot change ORB launch file. When I added my remap code to airsim_node.launch it does not work. The link that you have gave that says add remap link to subscriber's launch file which is ORB in our case. So, how can I change ORB launch file (stereo_inertial) ?  Actually, there is not any ".launch" file in ORB directories..

AnhTuDo1998, 2021-01-14:
Hi, may I know if you were able to run the demo? If yes, can you take a snapshot of ```rosrun rqt_graph rqt_graph``` for that?
Similarly, do the same when you trying to launch the orb_slam node and the camera node. Would help a lot for debugging

ryleymcc, 2021-04-02:
I'm wondering how I can make a launch file for ORB_SLAM3? rospack list does not find the ORB_SLAM3 package. I use this command to run the examples. rosrun ORB_SLAM3 Stereo path_to_vocabulary path_to_settings do_rectify