UZ-SLAMLab/ORB_SLAM3 #237: Why always report the error "Calibration parameters to rectify stereo are missing!" during run ORB_SLAM3 Stereo_Inertial?

yhy130531, 2021-01-12:
here, when I used real Sense T256  camera to run ORB_SLAM3 encounter a difficult question,that is,  the terminal always report " **_Calibration parameters to rectify stereo are missing_**!" during run ORB_SLAM3 Stereo_Inertial. I strictly according the command like "osrun ORB_SLAM3 Stereo_Inertial /home/haiyun/code/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/haiyun/code/ORB_SLAM3/Examples/my/TUM_512.yaml true" to run.

yhy130531, 2021-01-22:
Regarding this question, it is because before running with your own camera, you have to input the tpoic of your camera into the corresponding file. For example, my camera is Intel realsenseT265, and the topic corresponding to the image is: "/camera/fisheye1/image_raw" and "/Camera/fisheye2/image_raw".

closed 2021-01-22 by yhy130531

Maithilishetty, 2022-02-22:
Hi @yhy130531 I am facing the same issue while running the EuRoC bag file. Can you help me? I am not sure where I need to input the topic.

koksyuen, 2022-03-01:
@Maithilishetty I have faced same issue while running the EuRoC bag file. Did you managed to solve the issue?

Maithilishetty, 2022-03-02:
@koksyuen No I haven't been able to look at this for a while so have not fixed it yet. Will give it another shot and let you know if anything works! Please let me know in case you figure this out.

Maithilishetty, 2022-03-02:
@koksyuen I used the yaml file from Examples_old instead of Examples (i.e,  rosrun ORB_SLAM3 Stereo_Inertial Vocabulary/ORBvoc.txt Examples_old/Stereo-Inertial/EuRoC.yaml true). This had the rectification parameters. However, now I have another issue where the rosbag is not loading and I am getting a blank screen which says "waiting for images". Let me know if you run in the same issue and know how to fix it! Thanks.

koksyuen, 2022-03-14:
@Maithilishetty I didn't faced any problem after following your method

ShinChoon, 2022-04-04:
> Regarding this question, it is because before running with your own camera, you have to input the tpoic of your camera into the corresponding file. For example, my camera is Intel realsenseT265, and [...]

Hi may I ask shall the topic be "/camera/fisheye1/image_raw", the oroginal stereo_inertial has the topic as "/camera/left/image_raw". I am using T265 too, but I have troubles after VOC loaded. I want to know is that fisheye topic compulsory from bag file?

ShinChoon, 2022-04-04:
> Regarding this question, it is because before running with your own camera, you have to input the tpoic of your camera into the corresponding file. For example, my camera is Intel realsenseT265, and [...]

Hi may I ask shall the topic be

Hao-Starrr, 2022-05-13:
@Maithilishetty You may compare the two EuRoC.yaml. In Examples_old, it only has one camera parameter. I think that is the problem. 
And I copied the rectification parameter from old to new, it got "Fail to track local map!". Then I found it doesn't need rectification at all. If rectify, the images will retort, if not rectify, the images are perfect. So just run "rosrun ORB_SLAM3 Stereo_Inertial Vocabulary/ORBvoc.txt Examples_old/Stereo-Inertial/EuRoC.yaml false". Then I got the video.

Unknown9190, 2023-12-20:
it is the same problem when using the stereo, did anyone found a fix to this?