UZ-SLAMLab/ORB_SLAM3 #238: ot preintegrated measurement

roboticsai, 2021-01-16:
Whenever i'm trying to run the ros_mono_inertial node with image and imu topic being subscribed i'm getting this errros:
```
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 278
mnInitialFrameId = 208
3 Frames set to lost
First KF:4; Map init KF:2
New Map created with 204 points
Point distribution in KeyFrame: left-> 204 --- right-> 0
build optimization graph
Not preintegrated measurement
Segmentation fault (core dumped)
```

I'm using ubuntu 20.04 and ros noetic and have sucesfully built the orb_slam3 library and the ros_mono_inertial node.
The ros_mono node runs succesfully.

GabHoo, 2021-02-02:
Same but i am not using ROS. Moreover it gave me this error after few successful runs

kenmaro3, 2021-02-20:
@roboticsai @GabHoo 
Hi, I am reading the imu output from android device and got this error as well.
Any idea what is causing this yet?

rovinellimarco, 2021-05-22:
@roboticsai @GabHoo @kenmaro3 
Did any of you manage to solve the problem? I am getting the same error when trying to use the IMU measurements in a KITTI sequence.

FSRyan, 2021-08-10:
Hello!Have you solved this problem？

sandstorm666, 2022-05-05:
i am also encountering this problem,can anyone solve this problem?

MagdaZal, 2022-08-16:
I have the same problem, have you solved this?

> Empty IMU measurements vector!!!
> Not preintegrated measurement