UZ-SLAMLab/ORB_SLAM3 #243: Monocular scale error accumulation

yada222, 2021-01-21:
Hello everyone, I would like to ask a question. When I try to create images with a monocular camera（whether it’s a pinhole camera or a fisheye camera）, there will always be a situation where the scale error becomes larger and larger as the turns increase.
For example, the following is the test result of the public data set TUM_outdoor4. Although the tester is walking at a constant speed with the camera, the key frames and map points are all piled together。

[image]

My own cameras have also been calibrated and the calibration parameters have been modified。
I also tried some modifications, such as increasing the number of ORB points, filtering out orb points that are far away, etc., but the effect is not satisfactory.
Has anyone encountered a similar problem? Do you know the reason or find a solution?

CSU-NXY, 2021-03-24:
Same problem here. While test on KITTI sequence 01 with monocular mode, the scale suddenly changed.
[image]

treememory, 2021-06-12:
same problem. in same dataset, after multi scale optimization, scale tends to 1.  sometimes scale turn larger or samller.
sometimes, Converge to a const scale to real world. anyone fix it?  thank you~