UZ-SLAMLab/ORB_SLAM3 #245: When using the Intel realSense T265 camera to run ORB_SLAM3 Stereo + IMU under ROS, "not enough acceleration" will always appear. Has anyone encountered this problem?

yhy130531, 2021-01-22:
When using the Intel realSense T265 camera to run ORB_SLAM3 Stereo + IMU under ROS, "**not enough acceleration**" will always appear. Has anyone encountered this problem?
Below is the output of my terminal：
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Fisheye
- fx: 286.195
- fy: 286.1
- cx: 431.162
- cy: 405.501
- k1: -0.00388595
- k2: 0.0403609
- k3: -0.0385697
- k4: 0.00659577
- Camera1 Lapping: 0, 511

Camera2 Parameters:
- Camera: Fisheye
- fx: 285.214
- fy: 285.234
- cx: 418.29
- cy: 410.462
- k1: -0.00364469
- k2: 0.0366678
- k3: -0.0340943
- k4: 0.00507078
- mTlr: 
[0.999973, -0.00135725, -0.0071993899, -0.064281732;
 0.00134867, 0.99999797, -0.0011965201, -0.00016027065;
 0.0072010001, 0.00118678, 0.999973, -0.0001797156]
- Camera2 Lapping: 0, 511
- fps: 30
- color order: RGB (ignored if grayscale)

Depth Threshold (Close/Far Points): 2.69857

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[-0.99998403, 0.0056480002, 0.0002338, -0.053582013;
 -0.0056478502, -0.99998403, 0.000661169, -0.00014583991;
 0.00023753, 0.00065983803, 1, -0.00010266486;
 0, 0, 0, 1]

IMU frequency: 200 Hz
IMU gyro noise: 5e-06 rad/s/sqrt(Hz)
IMU gyro walk: 2.2e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 6e-05 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.00086 m/s^3/sqrt(Hz)
Discard points further than 20 m from current camera
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.not IMU meas
First KF:0; Map init KF:0
[... 543 lines cut ...]
Fail to track local map!
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 914
mnInitialFrameId = 912
637 Frames set to lost
not IMU meas
First KF:145; Map init KF:144
New Map created with 0 points
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 917
mnInitialFrameId = 915
638 Frames set to lost
not IMU meas

arashazimi0032, 2021-02-10:
hi @yhy130531 .
Has your problem been solved?
I have exactly the same problem, but in a different way.
I added a function to the Tracking.cc file but I have not yet called or used it anywhere in the program However, this error appears.
Does anyone know anything about this ?

yhy130531, 2021-02-10:
> hi @yhy130531 .
> Has your problem been solved?
> I have exactly the same problem, but in a different way.
> I added a function to the Tracking.cc file but I have not yet called or used it anywhere in the program However, this error appears.
> Does anyone know anything about this ?

hi@arashazimi0032，I have solved this problem，the problem occured because the update frequence desynchrony of acceleration and of that gyroscope  with Intel Realsense T265 camera, according to the Intel report, the acceleration frequence is 62Hz, but the gyroscope is 200hz. The solution is modify the contents in camera lunch file named "roslaunch realsense2_camera rs_t265.launch"，
<arg name="enable_sync"         default="true"/>
<arg name="linear_interpolation"    default="copy"/>
if you use the T265, I think this will help you, good luck!

of course, Before compiling ORB_SLAM3, change the camera TPOIC in the source code to the topic of your own camera, this point is very important!

closed 2021-02-10 by yhy130531