UZ-SLAMLab/ORB_SLAM3 #246: testing in ubuntu 20.04

Angit16, 2021-01-22:
Hello,

Has the current version been implemented in ubuntu 20.04?

thank you,
Anindya

rafa-chao, 2021-01-27:
Doesn't work due to OpenCV

Geogouz, 2021-01-27:
Which is the suggested operating system for running ORB-SLAM3?

rafa-chao, 2021-01-27:
Ubuntu 18.04.

Geogouz, 2021-01-31:
Are there any insights on what exactly the problem with OpenCV might be? Cause OpenCV compiles and runs ok (as standalone) on Ubuntu 20.04. How quickly may we expect for this problem to be addressed?

GabHoo, 2021-02-02:
I am running it  successfully  on ubuntu 20 with opencv 4.5.1 (I don't know precisely why this version of opencv but it does the job)

rafa-chao, 2021-02-03:
> I am running it successfully on ubuntu 20 with opencv 4.5.1 (I don't know precisely why this version of opencv but it does the job)

Hi Gabhoo. Nice done !!!!!
did you install opencv, or is the standard version that comes installed with ubuntu 20?

Are you using ROS as well?

rafa-chao, 2021-02-03:
> I am running it successfully on ubuntu 20 with opencv 4.5.1 (I don't know precisely why this version of opencv but it does the job)

I remember when I tried to install it in ubuntu 20 and opencv 4.2.0, that the orb_slam installation process failed due to opencv version was higher than 4.0. What modifications did you make in the installation process so orb_slam accepts opencv version?

Thanks !!!