UZ-SLAMLab/ORB_SLAM3 #248: Minimum IMU Hz?

jnatale11, 2021-01-22:
Hello, 
I'm seeing that the example datasets have IMU data at 200Hz, but I'm having trouble getting my [BNO055](https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor/overview) to even hit 100Hz. 

I'm wondering, is there some minimum IMU frequency that is expected for this system (or SLAM in general)? Some degradation in performance is alright for me, but not a ton.

Alternatively, does anyone have a recommendation for a better IMU to use?

peclatj, 2021-02-01:
Hello, I tested with the IMU data from the Kitti (raw,unsync) dataset at 100Hz and 10FPS, it works quite well.
[image]

I noticed that if you don't have at least one IMU measurement per frame, you get a SegFault.

SebsBarbas, 2021-02-02:
Hi @peclatj, as for the monocular configuration of sensors, for the calibration which values did you use or where did you find them? I have been looking and it seemed that they were not posted in the original dataset.

peclatj, 2021-02-05:
Hello, I found the calibration parameters on this page, http://www.cvlibs.net/datasets/kitti/raw_data.php
[image]
Here is the yaml file I use, but I only tested for stereo camera. Please ignore the field "tba" it's a custom parameter I added.
I used the synced+rectified images but unsync+unrectified for the IMU data.
[Kitti.zip](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/5934524/Kitti.zip)

closed 2021-02-10 by jnatale11

SebsBarbas, 2021-03-01:
Sorry for the late response @peclatj  I had not noticed your response. I was actually trying to refer to the noise of the IMU which is a parameter which acts as an input in: IMU.NoiseGyro, IMU.NoiseAcc etc.

peclatj, 2021-03-01:
Oh, I forgot about those, yeah I didn't found them either sorry. I just used the values from an IMU I own, because I forgot to check them. By the way, I just noticed that the IMU.Frequency is wrong in the file I sent, it should be 100. But it still works quite well.

Taoran2023, 2024-03-15:
hi, did you successfully run ORBSLAM3 with BNO055 IMU?