UZ-SLAMLab/ORB_SLAM3 #253: MapPoint handling after Loop Closures

SebsBarbas, 2021-01-29:
Hello everyone,
I was wondering how does ORB SLAM3 handles MapPoints when there is a Loop Closure. Lets say we have walked down a street and added all the observed MapPoints to the map. After a while we go around the same street, so the same key Features should be detected again. I was wondering if, as there would be the same MapPoints, if they are added again to the map. If it does not add them, could someone pin point me to the line of code where these get rejected? I have been trying to search for this and I am struggling a bit. Thanks in advance for all your help,
Sebastián

Giannis-Alamanos, 2021-02-27:
Hello, you should see LoopClosing.cc Lines 1155-1179.

SebsBarbas, 2021-03-01:
Hello Giannis, thank you  very much for your help. I will close now this thread.

closed 2021-03-01 by SebsBarbas