UZ-SLAMLab/ORB_SLAM3 #260: How to Close ORB-SLAM3?

Kanishk598, 2021-02-13:
I want to close ORB-SLAM3 by pressing CTRL+C, but it isn't closing. The Cross button on the Map Viewer window isn't working and the Current Frame window is getting reopened every time I'm closing it with the Cross button. And workaround?
Thanks

wzhwzh94, 2021-02-19:
Try like this:
https://github.com/NERanger/ORB-SLAM2-with-D435i/blob/master/Examples/RGB-D/rgbd_realsense.cc

Kanishk598, 2021-02-21:
> Try like this:
> https://github.com/NERanger/ORB-SLAM2-with-D435i/blob/master/Examples/RGB-D/rgbd_realsense.cc

Can you please tell me how am I supposed to use this code? 
I'm a beginner in ROS, any help is greatly appreciated 😃

Kanishk598, 2021-02-27:
Thank you, there was some temporary bug which is solved now. Pressing CTRL+C on the terminal running ORB is closing ORB3 without any difficulty or error. 
I am closing this issue.

closed 2021-02-27 by Kanishk598