UZ-SLAMLab/ORB_SLAM3 #262: symbol lookup error: /usr/lib/x86_64-linux-gnu/libGLX_mesa.so.0: undefined symbol: XGetXCBConnection

eduardoavila1m, 2021-02-25:
I am having issues compiling a mono image on my own dataset. The system was compiling properly now is shwing this error after runnign the following command. I can confirm that the png image is in the right directory ~Datasets/EuRoC/MH01

./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml "$pathDatasetEuroc"/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono
num_seq = 1
file name: dataset-MH01_mono
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 458.65399169921875
- fy: 457.29598999023438
- cx: 367.21499633789062
- cy: 248.375
- k1: -0.28340810537338257
- k2: 0.073959067463874817
- p1: 0.00019359000725671649
- p2: 1.7618711353861727e-05
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Failed to load image at: /MH01/mav0/cam0/data/1403636579763555584.png
./Examples/Monocular/mono_euroc: symbol lookup error: /usr/lib/x86_64-linux-gnu/libGLX_mesa.so.0: undefined symbol: XGetXCBConnection

I can not find steps to run ORB SLAM 3 on own data with out ROS. Would you mind helping me with this?

tbadave, 2022-11-01:
I'm currently having the same issue. I'm trying to run `stereo_euroc` with my custom dataset that mimics EuRoC dataset file structure.
Did you guys @eduardoavila1m @dframirez-usmc find a solution to this issue?

zarathacia, 2023-04-27:
You need to check the dataset path that you are providing