UZ-SLAMLab/ORB_SLAM3 #264: IMU infinity initialize problem

romaster93, 2021-02-26:
Input sensor was set to: Stereo-Inertial

Loading ORB Vocabulary. This could take a while...
[ INFO] [1614321981.095102525, 5584.707000000]: Stereo is NOT SUPPORTED
[ INFO] [1614321981.095151950, 5584.707000000]: OpenGl version: 4.6 (GLSL 4.6).
[ WARN] [1614321981.556999493, 5585.169000000]: No transform between frames /world and /base_link available after 5585.169000 seconds of waiting. This warning only prints once.
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 984.244
- fy: 980.814
- cx: 690
- cy: 233.197
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 10
- color order: RGB (ignored if grayscale)

Depth Threshold (Close/Far Points): 35.1879

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[0.014865543, -0.99988091, 0.004140297, 0;
 0.99955726, 0.014967213, 0.02571553, 0;
 -0.025774436, 0.0037561883, 0.99966073, 0;
 0, 0, 0, 1]

IMU frequency: 400 Hz
IMU gyro noise: 0.00017 rad/s/sqrt(Hz)
IMU gyro walk: 1.9393e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.002 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.003 m/s^3/sqrt(Hz)
not IMU meas
id last: 0    id curr: 1
Timestamp jump detected, before IMU initialization. Reseting...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 0
mnInitialFrameId = 0
0 Frames set to lost
not IMU meas
First KF:0; Map init KF:0
New Map created with 448 points
[ INFO] [1614321987.557391630, 5591.167000000]: Finished waiting for tf, waited 5591.167000 seconds
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
IMU in Map 0 is initialized
Not enough motion for initializing. Reseting...
TRACK: Reset map because local mapper set the bad imu flag 
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 3
mnInitialFrameId = 1
57 Frames set to lost
not IMU meas
First KF:11; Map init KF:0
New Map created with 466 points
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
IMU in Map 0 is initialized
Not enough motion for initializing. Reseting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
Not enough motion for initializing. Reseting...
TRACK: Reset map because local mapper set the bad imu flag 
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 62
mnInitialFrameId = 60
123 Frames set to lost
not IMU meas
First KF:23; Map init KF:11
New Map created with 453 points

IMU continues to be initialized and the position becomes 0,0,0. Also, the feature point on the image repeatedly flickers. While the drone is flying, the position changes abruptly and it cannot maintain its posture.

https://youtu.be/tU3YKnHOyks

Dose anyone know of a solution to this?

FredrikErikJohansson, 2021-03-02:
We are experiencing the same issue, did you solve it?

romaster93, 2021-03-03:
> We are experiencing the same issue, did you solve it?

no..... not yet......
However, there is a problem that requires initial movement.

ThornsOoO, 2021-03-12:
Some motion excitation is needed to initialize IMU

aravindSolteq, 2021-04-30:
Hi Did you find the solution for this. Im experiencing the same problem

WeilongXia, 2022-07-17:
@romaster93 Hi! Did you solved that? I'm experiencing the same problem, although the drone need initial movement, but it is not resonable that the position changes abruptly, is it? Any suggestions is appreciated.

wqrydqklyu, 2024-11-20:
I also have this problem. I did some initial excitations when I was collecting the data, but they still did not help. Has anybody solved it? I also get some output like this:
Not enough motion for initializing. Reseting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex