UZ-SLAMLab/ORB_SLAM3 #265: The timestamp problem of tum-vi dataset

AcresJoe1, 2021-03-07:
Thank you very much for your work. Why can't the timestamp of tum-vi dataset be printed out in the main function corresponding to the image name and the original timestamp？
The main function runs the command :   ./ examples / unitary / mono_ tum_ vi Vocabulary/ ORBvoc.txt Examples/Monocular/TUM_ 512.yaml  ./data/dataset-room1_ 512_ 16/mav0/cam0/data Examples/Monocular/TUM_ TimeStamps/dataset-room1_ 512.txt dataset-room1_ 512_ mono

num_seq = 1
file name: dataset-room1_512_mono
Loading images for sequence 0..../data/dataset-room1_512_16/mav0/cam0/data
Examples/Monocular/TUM_TimeStamps/dataset-room1_512.txt
LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Fisheye
- fx: 190.97846984863281
- fy: 190.97331237792969
- cx: 254.93170166015625
- cy: 256.89743041992188
- k1: 0.0034823894966393709
- k2: 0.00071503483923152089
- k3: -0.0020532361231744289
- k4: 0.00020293673151172698
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1500
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308199447552
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308249448704
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308299450624
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308349502464
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308399504640
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308449504512
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308499506432
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308549507584
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308599509504
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308649555456
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308699512576
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308749559040
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308799559936
tframe = vTimestampsCam[seq][ni]:---------tum_momo----- 1520530308849561856
...........

sandeepnmenon, 2022-03-01:
+1
I am observing the same thing. Is there any correction being done for tum_vi ?
@AcresJoe1 Did you find the issue?
I see there is a divising by 1e9 of the timestamp in the LoadImages code.
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/Examples/Monocular/mono_tum_vi.cc#L252

But this is done in EUROC as well.

sandeepnmenon, 2022-03-01:
I removed the division by 1e9 shown above and added the division to the waiting time since waiting time becomes scaled and the issue is resolved.

            // Wait to load the next frame
            double T=0;
            if(ni<nImages[seq]-1)
                T = vTimestampsCam[seq][ni+1]-tframe;
            else if(ni>0)
                T = tframe-vTimestampsCam[seq][ni-1];
            T/=1e9; // Added to reduce wait time
            if(ttrack<T)
                usleep((T-ttrack)*1e6); // 1e6