UZ-SLAMLab/ORB_SLAM3 #267: Stereo_Inertial initialization failed!

1216621137, 2021-03-10:
I am running Stereo_Inertial with the imu data and camera images from airsim, but I have encounter some problems:
When the kFs get 10, it will reset, and looping.
(following is the looping parts)
```
New Map created with 479 points
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
IMU in Map 0 is initialized
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 516
mnInitialFrameId = 514
550 Frames set to lost
not IMU meas
First KF:120; Map init KF:110
```

I have the similar issue before.
 I ran the stereo-inertial with ZED2 and it failed to initialization because of the wrong setting parameters(Tbc), but I noticed that  it didn't have "PIOLKF: NOT ENOUGH EDGES" error and it resetted when the "kFs" was 15.
This time, I assure that the parameters are accurate. So I have no idea about the real reason causing the issue. 
Hoping for your help!! Thank you!

1216621137, 2021-03-10:
https://user-images.githubusercontent.com/46966850/110624839-b20cc600-81d9-11eb-9c78-abcad9900495.mp4
This is the video.

ThornsOoO, 2021-03-12:
Some motion excitation is needed to initialize IMU

RoZhong, 2021-12-01:
> Some motion excitation is needed to initialize IMU

Can you describe it in detail, how do you do it?

ProVival, 2022-03-13:
> > Some motion excitation is needed to initialize IMU
> 
> Can you describe it in detail, how do you do it?

Were you able to solve the problem :/ I am having the same problem. I don't know if it is about 
```
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
```
or about imu not having enough excitation ( it says IMU in Map 0 is initialized after sometime so i don't know what to think :D)

ThornsOoO, 2022-03-13:
> Were you able to solve the problem :/ I am having the same problem. I don't know if it is about
>
> ```
> PIOLKF: NOT ENOUGH EDGES
> Fail to track local map!
> ```
>
> or about imu not having enough excitation [...]

At the beginning, I used the Euroc datasets without any problems. Later, after using my own device (inter d435i), I always had the problem of initialization failure, such as _fail to track local map!_ or _IMU exercise is insufficient_. Later, it was found that there were two reasons, one was the calibration problem between the camera and IMU, and the other was the motion state in the initialization process. In the case of handheld devices, the motion amplitude is small and the speed is close to a uniform speed. On the UAV and mobile platform, or in the case of handheld with larger movement range, which can be initialized normally

ProVival, 2022-03-13:
> At the beginning, I used the Euroc datasets without any problems. Later, after using my own device (inter d435i), I always had the problem of initialization failure, such as _fail to track local [...]

Thank you so much, i was trying to run it on Airsim. I used the same parameters i use on VINS-Fusion. Maybe i can try to run it on higher speeds. 

Do you have any idea about how to solve `PIOLKF: NOT ENOUGH EDGES`. Thanks so much again :D

ThornsOoO, 2022-03-14:
> > > > > Some motion excitation is needed to initialize IMU
> > > > 
> > > > 
> > > > Can you describe it in detail, how do you do it?
> > > 
> > > 
> > > Were you able to solve the problem :/ I am having the same problem. I don't know if it is about
> > > ```
> > > PIOLKF: NOT ENOUGH EDGES
> > > Fail to track local map!
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > or about imu not having enough excitation ( it says `IMU in Map 0 is initialized` after sometime so i don't know what to think :D)
> > 
> > 
> > At the beginning, I used the Euroc datasets without any problems. Later, after using my own device (inter d435i), I always had the problem of initialization failure, such as _fail to track local map!_ or _IMU exercise is insufficient_. Later, it was found that there were two reasons, one was the calibration problem between the camera and IMU, and the other was the motion state in the initialization process. In the case of handheld devices, the motion amplitude is small and the speed is close to a uniform speed. On the UAV and mobile platform, or in the case of handheld with larger movement range, which can be initialized normally
> 
> Thank you so much, i was trying to run it on Airsim. I used the same parameters i use on VINS-Fusion. Maybe i can try to run it on higher speeds.
> 
> Do you have any idea about how to solve `PIOLKF: NOT ENOUGH EDGES`. Thanks so much again :D
> 
> Edit: I will now try only stereo mode without imu Edit2: Okay, before i tried only stereo mode i wanted to make one last change. It was changing the parameter `ThDepth: 35.0` to `ThDepth: 50.0`. Now it started to work without always resetting the map.

I'm  sorry. I don't do this experiment now, so some questions are not clear. However, it should be noted that in the initialization process, the acceleration of motion should be obvious. For example, in the takeoff process of four rotor UAV, it does not mean that the faster the speed, the better. In addition, you need to move the camera up, down, left and right a little, so that ORBSLAM can capture more point cloud information in the map.

ProVival, 2022-03-14:
@ThornsOoO Thank you! I understand now

makolele12, 2022-03-30:
> At the beginning, I used the Euroc datasets without any problems. Later, after using my own device (inter d435i), I always had the problem of initialization failure, such as _fail to track local [...]

Hi, did you find a solution for the D435i? I am using the same camera and have the exact same problem with the initialization.

hplegend, 2024-05-04:
> because of the wrong setting parameters(Tbc)

I run ORB_SLAM3  using zed2 stereo image and imu, but it  was failed. I just want to known, Tbc is imu to camera0 or camera0 to imu？ I was pullzed by the ORB_SLAM3 source code, as the code cout a log "Left camera to Imu Transform (Tbc): " .