UZ-SLAMLab/ORB_SLAM3 #273: Difference between SaveTrajectoryEuRoC and SaveKeyFrameTrajectoryTUM?

ssemenova, 2021-03-22:
I'm wondering what the difference is between these two functions (and KeyFrameTrajectorytxt vs CameraTrajectory.txt). It looks like SaveTrajectoryEuRoC saves all keyframes across multiple maps and SaveKeyFrameTrajectoryTUM saves just one map, but why have this distinction? Which result should be used for evaluating ATE?

hellovuong, 2021-05-17:
In my opinion, `SaveKeyFrameTrajectory` only saves to the txt file the pose of `KeyFrame`. Meanwhile, `SaveTrajectory` saves to the file the `Frame's pose`. So, how they compute the `Frame's pose`? For each `Frame`, they stored the relative transformation between that `Frame` to its `reference KeyFrame`, then we can simply compute `Frame pose` by 
`Tcw (Frame pose) = relative transformation (mpTracker->mlRelativeFramePoses) *  reference KeyFrame's pose wrt the first KeyFrame (pKF->GetPose()*Two)`
Here is how they calculate the relative transformation between `Frame` and `Reference KeyFrame`:
`cv::Mat Tcr = mCurrentFrame.mTcw*mCurrentFrame.mpReferenceKF->GetPoseInverse();`
I think it is more objective if we use the SaveTrajectory for evaluating ATE.
Hope this can help and please correct me if I misunderstood!

Sherlock-hh, 2021-09-23:
> In my opinion, `SaveKeyFrameTrajectory` only saves to the txt file the pose of `KeyFrame`. Meanwhile, `SaveTrajectory` saves to the file the `Frame's pose`. So, how they compute the `Frame's pose`? [...]
Thanks for your anwer , but I still have question about the diffirence between relativeFramePoses and reference Keyframe , and I know there is a kind of frame named parentFrame , what the relationship between those kinds of frame? 
As I know, every keyframe pose has four Mat to record , like tcw, twc, ow, cw, the tcw and twc meas transformation matrix of camera coordinate and world coordinate. The World coordinate's original center should be the first init Frame's pose , so why not just save the Trajectory in World coordinate system(tcw* (1,1,1,1)), instead of use reference cameraFrame to calculate?