UZ-SLAMLab/ORB_SLAM3 #275: How to evaluate orbslam3？

AcresJoe1, 2021-03-22:
Hi,every one! I have a problem about orbslam3 evaluation, how to use Evo tool to evaluate monocular EUROC,  I have used f_ dataset-MH01_ mono.txt Compared with the ground_truth, it was found that the RMSE error was very large. I don't know whether it was the computer hardware or my evaluation method was wrong.
This is the result of my execution and evaluation：

 evo_ape tum data.tum  f_dataset-MH01_mono.txt -r full -va --plot --plot_mode xyz --save_plot ./APE_plot --save_results ./APE.zip
--------------------------------------------------------------------------------
Loaded 36382 stamps and poses from: data.tum
Loaded 3663 stamps and poses from: f_dataset-MH01_mono.txt
Synchronizing trajectories...
Found 3638 of max. 3663 possible matching timestamps between...
	data.tum
and:	f_dataset-MH01_mono.txt
..with max. time diff.: 0.01 (s) and time offset: 0.0 (s).
--------------------------------------------------------------------------------
Aligning using Umeyama's method...
Rotation of alignment:
[[ 0.33943501  0.28341718 -0.89691615]
 [ 0.93974145 -0.14360177  0.31026528]
 [-0.04086424 -0.94818418 -0.31508233]]
Translation of alignment:
[2.77015796 1.64419051 0.42545537]
Scale correction: 1.0
--------------------------------------------------------------------------------
Compared 3638 absolute pose pairs.
Calculating APE for full transformation pose relation...
--------------------------------------------------------------------------------
APE w.r.t. full transformation (unit-less)
(with SE(3) Umeyama alignment)

       max	6.060663
      mean	3.699845
    median	3.915156
       min	2.122828
      rmse	3.839457
       sse	53629.330044
       std	1.025954

--------------------------------------------------------------------------------
Plotting results... 
Plot saved to ./APE_plot_raw
Plot saved to ./APE_plot_map
--------------------------------------------------------------------------------
Saving results to ./APE.zip...

AcresJoe1, 2021-06-24:
> you should add the option `-as` to correct the scale of KeyFrameTrajectory, it should be like this:
> `evo_ape tum data.tum f_dataset-MH01_mono.txt -r full -va -as -p --plot_mode xyz --save_plot ./APE_plot --save_results ./APE.zip`

Thank you for your reply！

SandraKengne, 2025-05-16:
Please how to evaluate his performance?  I am having trouble obtaining the ATE RMSE values for ORBSLAM3 when I compile 10 times and take the median. They are always higher than what is stated in the literature. I would like to know if there are any parameters to modify in the code before compilation. Thank you in advance for your response.