UZ-SLAMLab/ORB_SLAM3 #277: Monocular+Inertial problem-Waiting for images

mmahdavian, 2021-03-24:
Hi.I am trying to use EuRoC dataset and I have an issue when I use Monocular+ inertial orb-slam3. The algorithm doesn't see images and only shows "Waiting for Images". Also, I get no errors. I have tested it with Monocular and it's OK.But with Inertial it doesn't work. I have tested it with different datasets and the same problem happens. This is the command that I use to run ORB-SLAM3:

rosrun ORB_SLAM3 Mono_Inertial Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/EuRoC.yaml

This is the output that I get:
```
Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
K is: [458.65399, 0, 367.215;
 0, 457.29599, 248.375;
 0, 0, 1]

Camera Parameters: 
- fx: 458.654
- fy: 457.296
- cx: 367.215
- cy: 248.375
- bf: 0
- k1: -0.283408
- k2: 0.0739591
- p1: 0.00019359
- p2: 1.76187e-05
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[0.014865543, -0.99988091, 0.004140297, -0.021640146;
 0.99955726, 0.014967213, 0.02571553, -0.064676985;
 -0.025774436, 0.0037561883, 0.99966073, 0.0098107308;
 0, 0, 0, 1]

IMU frequency: 200 Hz
IMU gyro noise: 0.00017 rad/s/sqrt(Hz)
IMU gyro walk: 1.9393e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.002 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.003 m/s^3/sqrt(Hz)
```
[image]

sumitsarkar1, 2021-03-28:
play your rosbag file used for MOno-imu and check the topics published with rostopic `list.../imu` topic should also be published

mmahdavian, 2021-03-28:
@sumitsarkar1 both are getting published

sumitsarkar1, 2021-03-28:
1.) rosbag play --pause MH03.bag /cam0/image_raw:=/camera/image_raw /imu0:=/imu [from here](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets#available_data)
2.) inside ORBSLAM3 dir .... : rosrun ORB_SLAM3 Mono_Inertial Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/EuRoC.yaml true

mmahdavian, 2021-03-28:
@sumitsarkar1 It didn't work. I am publishing data on /imu and /camera/image_raw. But the ORB-SLAM3 doesn't start. The weird thing is, on Stereo-Inertial the orb-slam3 starts working(although it can't initialize) but on Monocular-Inertial it doesn't even show images.

sumitsarkar1, 2021-03-29:
which rosbag are you using? if you can share the link I can have a look at them ( stereo IMU and mono IMU)

mmahdavian, 2021-03-29:
Thanks @sumitsarkar1 I tried with the link that you sent (from here Machine Hall 03) and I used "MH_03_medium.bag" and it didn't work on Monocular-Inertial. It's link is:
http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/machine_hall/MH_03_medium/MH_03_medium.bag

Thanks

sumitsarkar1, 2021-03-29:
check with rviz if you can see the images published by the bag file....and do a rostopic echo /imu to check imu...but rviz check is important...let me know

mmahdavian, 2021-03-30:
Thanks @sumitsarkar1 . I tried it in rviz. I see the images, but the imu has an error:
For frame [imu4]: Fixed Frame [world] does not exist. I don't know why do I have this error. I have "world" in fixed frame in global options. But still I get this error. Do you know how should I fix it?

Thank You

[image]

mmahdavian, 2021-04-02:
sorry it was my fault. I was making changes to ros_mono_inertial.cc code which caused the problem.

closed 2021-04-02 by mmahdavian

Maithilishetty, 2022-03-03:
@sumitsarkar1 @mmahdavian I am having the same issue, I am obtaining waiting for images regardless of the setting I am running it in for the EuRoC dataset. Could you please help me with this?

foemre, 2022-08-18:
I am also getting the "Waiting for images" message when running Mono-Inertial node, any solutions?

Maithilishetty, 2022-08-19:
Hi, if i remember correctly - it was because I was not using the right topic name that was subscribed within the code. I think you should check that.

LarryDong, 2023-03-26:
"Waiting for image" window will only show images when matches are found.
However, for me, the mono-imu mode cannot always extracts enough matches, so the window is always black.
I think we should check the calibration & match process.

Moynstain, 2024-01-30:
> "Waiting for image" window will only show images when matches are found. However, for me, the mono-imu mode cannot always extracts enough matches, so the window is always black. I think we should check the calibration & match process.

Thanks! I was experiencing the same issue with a normally-functional previous trial while appearing this "waiting for image" issue in the next trial. I tried to rotate the camera slightly and it works fine now :)

zxytql, 2024-03-14:
Thanks @Moynstain. I have solved it with your comment. I tried to change my camera resolution to 640 x 480 but it also shows "Waiting for image". But it works fine and shows image after I move and rotate my camera.

Redhwan-A, 2024-11-20:
@zxytql thank you so much. Move and rotate my camera is working with me.