UZ-SLAMLab/ORB_SLAM3 #278: the problem  ./build.sh

987428377, 2021-03-27:
ubuntu 18.04 

when I input     the last step of ./build.sh  :
   echo "Configuring and building ORB_SLAM3 ..."
   mkdir build
   cd build
   cmake .. -DCMAKE_BUILD_TYPE=Release
  make -j

there are no error but 
     CMakeFiles/Makefile2:289: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
     make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
     Makefile:83: recipe for target 'all' failed
     make: *** [all] Error 2

and all make output as follows:
 [ 69%] Linking CXX shared library ../../../Thirdparty/g2o/lib/libg2o.so
[ 69%] Built target g2o
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc: In member function ‘bool ORB_SLAM3::Tracking::ParseORBParamFile(cv::FileStorage&)’:
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:774:100: warning: ‘nLevels’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         mpIniORBextractor = new ORBextractor(5*nFeatures,fScaleFactor,nLevels,fIniThFAST,fMinThFAST);
                                                                                                    ^
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:774:100: warning: ‘fScaleFactor’ may be used uninitialized in this function [-Wmaybe-uninitialized]
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:777:41: warning: ‘nFeatures’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     cout << "- Number of Features: " << nFeatures << endl;
                                         ^~~~~~~~~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc: In member function ‘bool ORB_SLAM3::Tracking::ParseIMUParamFile(cv::FileStorage&)’:
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:881:52: warning: ‘Ngw’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     mpImuCalib = new IMU::Calib(Tbc,Ng*sf,Na*sf,Ngw/sf,Naw/sf);
                                                 ~~~^~~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:881:45: warning: ‘Na’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     mpImuCalib = new IMU::Calib(Tbc,Ng*sf,Na*sf,Ngw/sf,Naw/sf);
                                           ~~^~~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:881:39: warning: ‘Ng’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     mpImuCalib = new IMU::Calib(Tbc,Ng*sf,Na*sf,Ngw/sf,Naw/sf);
                                     ~~^~~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:811:11: warning: ‘freq’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     float freq, Ng, Na, Ngw, Naw;
           ^~~~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc: In member function ‘bool ORB_SLAM3::Tracking::ParseCamParamFile(cv::FileStorage&)’:
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:314:27: warning: ‘k4’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float k1, k2, k3, k4;
                           ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:314:23: warning: ‘k3’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float k1, k2, k3, k4;
                       ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:314:19: warning: ‘k2’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float k1, k2, k3, k4;
                   ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:314:15: warning: ‘k1’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float k1, k2, k3, k4;
               ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:313:27: warning: ‘cy’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                           ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:313:23: warning: ‘cx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                       ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:313:19: warning: ‘fy’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                   ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:313:15: warning: ‘fx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
               ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:178:27: warning: ‘cy’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                           ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:178:23: warning: ‘cx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                       ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:178:19: warning: ‘fy’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                   ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:178:15: warning: ‘fx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
               ^~
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc: In member function ‘void ORB_SLAM3::Tracking::Track()’:
/home/zhaofalong/ORB_SLAM3/src/Tracking.cc:1711:13: warning: ‘bOK’ may be used uninitialized in this function [-Wmaybe-uninitialized]
             if(bOK)
             ^~
In file included from /home/zhaofalong/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:38:0,
                 from /home/zhaofalong/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/zhaofalong/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/zhaofalong/ORB_SLAM3/include/LoopClosing.h:34,
                 from /home/zhaofalong/ORB_SLAM3/include/Optimizer.h:26,
                 from /home/zhaofalong/ORB_SLAM3/src/Optimizer.cc:20:
/home/zhaofalong/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/hyper_graph.h: In static member function ‘static void ORB_SLAM3::Optimizer::FullInertialBA(ORB_SLAM3::Map*, int, bool, long unsigned int, bool*, bool, float, float, Eigen::VectorXd*, bool*)’:
/home/zhaofalong/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/hyper_graph.h:148:116: warning: ‘VA2’ may be used uninitialized in this function [-Wmaybe-uninitialized]
           void setVertex(size_t i, Vertex* v) { assert(i < _vertices.size() && "index out of bounds"); _vertices[i]=v;}
/home/zhaofalong/ORB_SLAM3/src/Optimizer.cc:526:42: note: ‘VA2’ was declared here
                 g2o::HyperGraph::Vertex* VA2;
                                          ^~~
In file included from /home/zhaofalong/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:38:0,
                 from /home/zhaofalong/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/zhaofalong/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/zhaofalong/ORB_SLAM3/include/LoopClosing.h:34,
                 from /home/zhaofalong/ORB_SLAM3/include/Optimizer.h:26,
                 from /home/zhaofalong/ORB_SLAM3/src/Optimizer.cc:20:
/home/zhaofalong/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/hyper_graph.h:148:116: warning: ‘VG2’ may be used uninitialized in this function [-Wmaybe-uninitialized]
           void setVertex(size_t i, Vertex* v) { assert(i < _vertices.size() && "index out of bounds"); _vertices[i]=v;}
/home/zhaofalong/ORB_SLAM3/src/Optimizer.cc:525:42: note: ‘VG2’ was declared here
                 g2o::HyperGraph::Vertex* VG2;
                                          ^~~
/home/zhaofalong/ORB_SLAM3/src/Optimizer.cc:490:53: warning: ‘pIncKF’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         VertexAccBias* VA = new VertexAccBias(pIncKF);
                                                     ^
CMakeFiles/Makefile2:289: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
 

and then ,I try again,

[ 37%] Built target g2o
make[2]: *** No rule to make target '/opt/ros/melodic/lib/librealsense2.so', needed by '../lib/libORB_SLAM3.so'. Stop.
CMakeFiles/Makefile2:289: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Who can help me ? Thanks!

sumitsarkar1, 2021-03-28:
in the last line of build.sh ..try doing just make instead of make -j....not sure whether it will solve your problem

987428377, 2021-03-28:
> in the last line of build.sh ..try doing just make instead of make -j....not sure whether it will solve your problem

Thank you! I tried，but，it did not work.

[ 32%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/TwoViewReconstruction.cc.o
make[2]: *** No rule to make target '/opt/ros/melodic/lib/librealsense2.so', needed by '../lib/libORB_SLAM3.so'. Stop.
CMakeFiles/Makefile2:289: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

sumitsarkar1, 2021-03-28:
Is DBOW2, g2o and Vocab getting build successfully ..? I am not understanding why librealsense2.so is in the error..its neither required by ROS melodic nor ORBSLAM3...hope you have not tampered with the CMakeLists.txt of ORB SLAM3...

987428377, 2021-03-29:
> Is DBOW2, g2o and Vocab getting build successfully ..? I am not understanding why librealsense2.so is in the error..its neither required by ROS melodic nor ORBSLAM3...hope you have not tampered with the CMakeLists.txt of ORB SLAM3...

Thanks for your reply！I tried to search for librealsense2.so in all the files and copy it to'/opt/ros/melodic/lib/' and the compilation was successful

closed 2021-04-06 by 987428377