UZ-SLAMLab/ORB_SLAM3 #279: build_ros.sh issue: 2 versions of libopencv are needed and conflicted

Jackyxtt, 2021-03-28:
CMakeLists find_package(opencv3), error occurred like this: 

[ 15%] Linking CXX executable ../Stereo_Inertial
/usr/bin/ld: warning: libopencv_imgproc.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_imgproc.so.3.4
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_core.so.3.4
/usr/bin/ld: CMakeFiles/Stereo_Inertial.dir/src/ros_stereo_inertial.cc.o: undefined reference to symbol '_ZNK2cv8FileNodecviEv'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libopencv_core.so.4.2.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

I changed the CMakeLists to find_package(opencv4),
[ 15%] Linking CXX executable ../Stereo_Inertial
/usr/bin/ld: warning: libopencv_highgui.so.3.4, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_highgui.so.4.2
/usr/bin/ld: warning: libopencv_calib3d.so.3.4, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_calib3d.so.4.2
/usr/bin/ld: warning: libopencv_imgproc.so.3.4, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_imgproc.so.4.2
/usr/bin/ld: warning: libopencv_core.so.3.4, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_core.so.4.2
afterwards every Linking process has warning like this.

Different versions libopencv shared libs are needed and inevitably conflict with another version.

sumitsarkar1, 2021-03-28:
Different versions are not needed...the version ROS uses and ORBSLAM3 ises are not the same...I guess if you are not building ROS from source you dont have control over the version of OpenCV it uses....so build ORBSLAM3 with the version of OpenCV your ROS uses.....in your case ROS noetic uses Opencv 4.2 and you have build ORBSLAM3 with OpenCV 3.4 . Try to use OpenCV 4.2 with ORBSLAM3

Jackyxtt, 2021-03-29:
Problem solved, thanks a lot

aprotyas, 2021-04-12:
@Jackyxtt were you able to get ORBSLAM3 to build with OpenCV 4.2?

Jackyxtt, 2021-04-13:
> @Jackyxtt were you able to get ORBSLAM3 to build with OpenCV 4.2?

Yes, just change the CMakeLists to find_package(OpenCV 4) and add #include "opencv2/imgcodecs/legacy/constants_c.h" in all .cc file in Examples. It will work

ryleymcc, 2021-04-16:
Please close the issue if your question has been answered.

closed 2021-04-16 by Jackyxtt