UZ-SLAMLab/ORB_SLAM3 #280: run orb-slam3 with D435i

987428377, 2021-03-29:
When I input   rosrun
 ORB_SLAM3 RGBD /home/***/ORB_SLAM3/Vocabulary/ORBvoc.txt  /home/***/ORB_SLAM3/Examples/RGB-D/d435i.yaml
occurred
/home/***/ORB_SLAM3/Examples/ROS/ORB_SLAM3/RGBD: error while loading shared libraries: librealsense2.so.2.42: cannot open shared object file: No such file or directory
Where is librealsense 2.so.2.42 loaded from？ Thanks！

sumitsarkar1, 2021-03-29:
1.) D435I needs librealsense2 SDK to function [here](https://github.com/IntelRealSense/librealsense). You can build it from source or install pre-build libs. This will install the librealsense2.so on your system. If your build from source it will be in /usr/local/lib
2.) If using ROS one has to use the ROS wrapper [here](https://github.com/IntelRealSense/realsense-ros). Use Catkin_make for installing
3 .) Once 2 is done you should be able to launch D435I : roslaunch realsense2_camera rs_rgbd.launch
4.) Start ORBSLAM3 using the D435I.yaml. The present ORBSLAM3 repo does not have D435i.yaml. Make sure the parameters are correct and in TUM format

surfii3z, 2021-04-01:
> Where is librealsense 2.so.2.42 loaded from？ Thanks！

I think this problem could be solved by rebuilding the Pangolin from source again, then ORB_SLAM3 again.

surfii3z, 2021-04-01:
> /home/***/ORB_SLAM3/Examples/ROS/ORB_SLAM3/RGBD: error while loading shared libraries: librealsense2.so.2.42: cannot open shared object file: No such file or directory
I think this problem could be solved by rebuilding the Pangolin from source again, then ORB_SLAM3 again. I guess the problem is because Pangolin was compile with librealsense 2.42, but then your current librealsense got updated so it's broken.

987428377, 2021-04-02:
> I think this problem could be solved by rebuilding the Pangolin from source again, then ORB_SLAM3 again. I guess the problem is because Pangolin was compile with librealsense 2.42, but then your [...]
ok I will try your suggestion !Thanks!

closed 2021-04-06 by 987428377

celikemir, 2023-10-19:
> I think this problem could be solved by rebuilding the Pangolin from source again, then ORB_SLAM3 again. I guess the problem is because Pangolin was compile with librealsense 2.42, but then your [...]

You are correct ! this is exactly come from pangolin. If you install librealsense ros dependency you should delete and only use sdk provided .so 's.  Delete build directory then re-build.