UZ-SLAMLab/ORB_SLAM3 #282: run the ros Stereo_inertial(in ROS) blocked with "not IMU meas"?

Seasons2020, 2021-03-31:
When I run the Stereo_iertial in ROS，I get the Information about issure

    rosrun ORB_SLAM3 Stereo_Inertial Vocabulary/ORBvoc.txt MyConfiguation/MyStereo_Inertial.yaml false
    
    ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
    ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
    This program comes with ABSOLUTELY NO WARRANTY;
    This is free software, and you are welcome to redistribute it
    under certain conditions. See LICENSE.txt.
    
    Input sensor was set to: Stereo-Inertial
    
    Loading ORB Vocabulary. This could take a while...
    Vocabulary loaded!
    
    Creation of new map with id: 0
    Creation of new map with last KF id: 0
    Seq. Name: 
    
    Camera Parameters: 
    - Camera: Pinhole
    - fx: 609.078
    - fy: 609.097
    - cx: 323.715
    - cy: 239.456
    - k1: 0
    - k2: 0
    - p1: 0
    - p2: 0
    - fps: 30
    - color order: RGB (ignored if grayscale)
    
    Depth Threshold (Close/Far Points): 2.8731
    
    ORB Extractor Parameters: 
    - Number of Features: 1200
    - Scale Levels: 8
    - Scale Factor: 1.2
    - Initial Fast Threshold: 20
    - Minimum Fast Threshold: 7
    
    Left camera to Imu Transform (Tbc): 
    [1, 0, 0, -0.0055200001;
     0, 1, 0, 0.0051000002;
     0, 0, 1, 0.01174;
     0, 0, 0, 1]
    
    IMU frequency: 400 Hz
    IMU gyro noise: 0.0032101 rad/s/sqrt(Hz)
    IMU gyro walk: 2.24532e-05 rad/s^2/sqrt(Hz)
    IMU accelerometer noise: 0.0223046 m/s^2/sqrt(Hz)
    IMU accelerometer walk: 0.000418772 m/s^3/sqrt(Hz)
    not IMU meas
    not IMU meas
    not IMU meas
    not IMU meas
    not IMU meas
    not IMU meas

**I have try many methods that I can try, such as #620, #28, #150 , #125, #267 which are related to the issue about "_not IMU meas_", but my issue also exist.Maybe the issue is same with #132. Because it work when I run Stereo.I also use realsense camera D435i. I alse guess that the issue resulted from my wrong IMU initialization, then I do same things as #150,   tilting the system in pitch and roll at the beginning. But all obove don't work well.
I guess that there is something wrong with my yaml file.**

The following is my configuration:

      %YAML:1.0
      
      Camera.type: "PinHole"
      
      Camera.fx: 609.077697753906
      Camera.fy: 609.097290039062
      Camera.cx: 323.714874267578
      Camera.cy: 239.456405639648
      
      Camera.k1: 0.0
      Camera.k2: 0.0
      Camera.p1: 0.0
      Camera.p2: 0.0
      
      Camera.bFishEye: 0
      
      Camera.width: 640
      Camera.height: 480
      
      Camera.fps: 30.0
      
      Camera.bf: 49.9983802437782
      
      # Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
      Camera.RGB: 1
      
      # Close/Far threshold. Baseline times.
      ThDepth: 35.0
      
      # 矫正参数
      # Transformation from camera 0(infra1) to body-frame (imu)
      Tbc: !!opencv-matrix
         rows: 4
         cols: 4
         dt: f
         data: [1.0, 0.0, 0.0, -0.00552000012248755, 
                  0.0, 1.0, 0.0, 0.00510000018402934,
                  0.0, 0.0, 1.0, 0.011739999987185,
                  0.0,0.0,0.0,1.0]
      
      # IMU noise
      IMU.NoiseGyro: 3.2100952823795516e-03 # 1.6968e-04 
      IMU.NoiseAcc: 2.2304645905508961e-02 # 2.0000e-3
      IMU.GyroWalk: 2.2453169929039468e-05 
      IMU.AccWalk: 4.1877218867360575e-04 # 3.0000e-3
      IMU.Frequency: 400
      
      #--------------------------------------------------------------------------------------------
      # ORB Parameters
      #--------------------------------------------------------------------------------------------
      
      # ORB Extractor: Number of features per image
      ORBextractor.nFeatures: 1200
      
      # ORB Extractor: Scale factor between levels in the scale pyramid 	
      ORBextractor.scaleFactor: 1.2
      
      # ORB Extractor: Number of levels in the scale pyramid	
      ORBextractor.nLevels: 8
      
      # ORB Extractor: Fast threshold
      # Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
      # Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
      # You can lower these values if your images have low contrast			
      ORBextractor.iniThFAST: 20
      ORBextractor.minThFAST: 7
      
      #--------------------------------------------------------------------------------------------
      # Viewer Parameters
      #--------------------------------------------------------------------------------------------
      Viewer.KeyFrameSize: 0.05
      Viewer.KeyFrameLineWidth: 1
      Viewer.GraphLineWidth: 0.9
      Viewer.PointSize: 2
      Viewer.CameraSize: 0.08
      Viewer.CameraLineWidth: 3
      Viewer.ViewpointX: 0
      Viewer.ViewpointY: -0.7
      Viewer.ViewpointZ: -1.8
      Viewer.ViewpointF: 500

The "Tbc" is provided with by D435i, and it is the Extrinsic from "Infrared 1" To  "Imu" , as well as Intrinsic of camera. I use `rs-enumerate-devices -c` get device information about D435i.
Can you give me some advices to solve it?

Seasons2020, 2021-04-07:
This issue was caused by the version of realsense_ros.

closed 2021-04-07 by Seasons2020

hankge, 2023-08-14:
Hi, I meet the same error when I am using my own data to run stereo_inertial, do you how to fix it?