UZ-SLAMLab/ORB_SLAM3 #284: Implementation with D435 and T265 INTEL cameras of STEREO-INERTIAL

Shkunn, 2021-04-01:
## Hi everyone, 

I've been trying to implement ORB SLAM3 for some time now with the D435 cameras for their stereo and the T265 for its internal IMU. 

### Unfortunately I am stuck with the following error: 
> IMU in Map 0 is initialized
> Not enough motion for initializing. Reseting...
> TRACK: Reset map because local mapper set the bad imu flag 
> LM: Active map reset recieved
> LM: Active map reset, waiting...
> LM: Reseting current map in Local Mapping...
> LM: End reseting Local Mapping...
> LM: Reset free the mutex
> LM: Active map reset, Done!!!
> mnFirstFrameId = 1360
> mnInitialFrameId = 1359
> 1406 Frames set to lost
> First KF:221; Map init KF:210
> New Map created with 271 points
> build optimization graph
> start optimization
> end optimization
> update Keyframes velocities and biases
> IMU in Map 0 is initialized

### I can also see the error on what ORB SLAM3 shows me since after 10 KeyFrames the map resets and starts again in a loop...

By the way, I did a lot of research on the internet before coming here to ask my question. Some people talk about a problem of coordinates especially in the rotation between the IMU and the camera but INTEL gives a rotation matrix and a 3D printable base. So I don't know what to think. You could check this right here by clicking on the H_t265_d400.cfg file: https://github.com/IntelRealSense/librealsense/tree/master/examples/tracking-and-depth or going on this website : https://dev.intelrealsense.com/docs/depth-and-tracking-cameras-alignment

### I am looking for an answer from you! Hope you guys could help me!

thomasraynal98, 2021-04-02:
I have the same problem ! If you got the answer i'm very interested !

WaterHyacinthInNANHU, 2021-04-08:
Having the same problem with Mono_Inertial!

antithing, 2021-04-10:
Same here! If you succeed with the t265 sensor please share your code! Thanks

ryleymcc, 2021-04-10:
Can someone please post a yaml file for t265. I'm trying to get stereo inertial working but I cant find info on calibration.

Shkunn, 2021-04-10:
> Same here! If you succeed with the t265 sensor please share your code! Thanks

Do you mean with Stereo Inertial or only the Stereo mode with the T265 ?

Shkunn, 2021-04-10:
> Can someone please post a yaml file for t265. I'm trying to get stereo inertial working but I cant find info on calibration.

Hi ! Yes of course, here is the yaml Im usine for the T265 : 
`
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 286.304595947266
Camera.fy: 286.273712158203
Camera.cx: 417.1953125
Camera.cy: 399.115386962891

Camera.k1: -0.00604297406971455
Camera.k2: 0.0451732017099857
Camera.p1: -0.0425471812486649
Camera.p2: 0.00776465190574527

Camera.bFishEye: 100

Camera.width: 848
Camera.height: 800

# Camera frames per second 
Camera.fps: 30.0

# stereo baseline times fx
Camera.bf: 64.0611

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 35.0

#--------------------------------------------------------------------------------------------
# Stereo Rectification. Only if you need to pre-rectify the images.
# Camera.fx, .fy, etc must be the same as in LEFT.P
#--------------------------------------------------------------------------------------------
LEFT.height: 800
LEFT.width: 848
LEFT.D: !!opencv-matrix
   rows: 1
   cols: 5
   dt: d
   data:[-0.00604297406971455, 0.0451732017099857, -0.0425471812486649, 0.00776465190574527, 0]
LEFT.K: !!opencv-matrix
   rows: 3
   cols: 3
   dt: ds
   data: [286.3045, 0.0, 417.1953, 0.0, 286.2737, 399.1153, 0.0, 0.0, 1.0]
LEFT.R:  !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [0.999966347530033, -0.001422739138722922, 0.008079580483432283, 0.001365741834644127, 0.9999741760894847, 0.007055629199258132, -0.008089410156878961, -0.007044357138835809, 0.9999424675829176]
LEFT.P:  !!opencv-matrix
   rows: 3
   cols: 4
   dt: d
   data: [435.2046959714599, 0, 367.4517211914062, 0,  0, 435.2046959714599, 252.2008514404297, 0,  0, 0, 1, 0]

RIGHT.height: 800
RIGHT.width: 848
RIGHT.D: !!opencv-matrix
   rows: 1
   cols: 5
   dt: d
   data:[-0.00477181980386376, 0.0391876511275768, -0.0359093882143497, 0.00556322885677218, 0]
RIGHT.K: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [286.6798, 0.0, 422.400, 0.0, 286.5119, 398.7724, 0.0, 0.0, 1]
RIGHT.R:  !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [0.9999633526194376, -0.003625811871560086, 0.007755443660172947, 0.003680398547259526, 0.9999684752771629, -0.007035845251224894, -0.007729688520722713, 0.007064130529506649, 0.999945173484644]
RIGHT.P:  !!opencv-matrix
   rows: 3
   cols: 4
   dt: d
   data: [435.2046959714599, 0, 367.4517211914062, -47.90639384423901, 0, 435.2046959714599, 252.2008514404297, 0, 0, 0, 1, 0]

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1800

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.8

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 12

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500
`

Shkunn, 2021-04-10:
> Can someone please post a yaml file for t265. I'm trying to get stereo inertial working but I cant find info on calibration.

Yes of course ! This is the one I use for Stereo only with the T265 : 

%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 286.304595947266
Camera.fy: 286.273712158203
Camera.cx: 417.1953125
Camera.cy: 399.115386962891

Camera.k1: -0.00604297406971455
Camera.k2: 0.0451732017099857
Camera.p1: -0.0425471812486649
Camera.p2: 0.00776465190574527

Camera.bFishEye: 100

Camera.width: 848
Camera.height: 800

# Camera frames per second 
Camera.fps: 30.0

# stereo baseline times fx
Camera.bf: 64.0611

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 35.0

#--------------------------------------------------------------------------------------------
# Stereo Rectification. Only if you need to pre-rectify the images.
# Camera.fx, .fy, etc must be the same as in LEFT.P
#--------------------------------------------------------------------------------------------
LEFT.height: 800
LEFT.width: 848
LEFT.D: !!opencv-matrix
   rows: 1
   cols: 5
   dt: d
   data:[-0.00604297406971455, 0.0451732017099857, -0.0425471812486649, 0.00776465190574527, 0]
LEFT.K: !!opencv-matrix
   rows: 3
   cols: 3
   dt: ds
   data: [286.3045, 0.0, 417.1953, 0.0, 286.2737, 399.1153, 0.0, 0.0, 1.0]
LEFT.R:  !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [0.999966347530033, -0.001422739138722922, 0.008079580483432283, 0.001365741834644127, 0.9999741760894847, 0.007055629199258132, -0.008089410156878961, -0.007044357138835809, 0.9999424675829176]
LEFT.P:  !!opencv-matrix
   rows: 3
   cols: 4
   dt: d
   data: [435.2046959714599, 0, 367.4517211914062, 0,  0, 435.2046959714599, 252.2008514404297, 0,  0, 0, 1, 0]

RIGHT.height: 800
RIGHT.width: 848
RIGHT.D: !!opencv-matrix
   rows: 1
   cols: 5
   dt: d
   data:[-0.00477181980386376, 0.0391876511275768, -0.0359093882143497, 0.00556322885677218, 0]
RIGHT.K: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [286.6798, 0.0, 422.400, 0.0, 286.5119, 398.7724, 0.0, 0.0, 1]
RIGHT.R:  !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [0.9999633526194376, -0.003625811871560086, 0.007755443660172947, 0.003680398547259526, 0.9999684752771629, -0.007035845251224894, -0.007729688520722713, 0.007064130529506649, 0.999945173484644]
RIGHT.P:  !!opencv-matrix
   rows: 3
   cols: 4
   dt: d
   data: [435.2046959714599, 0, 367.4517211914062, -47.90639384423901, 0, 435.2046959714599, 252.2008514404297, 0, 0, 0, 1, 0]

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1800

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.8

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 12

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

ryleymcc, 2021-04-10:
what version of opencv are you using?

ryleymcc, 2021-04-10:
what about for stereo inertial?

Shkunn, 2021-04-10:
> what about for stereo inertial?

For stereo inertial I used the D435 for the stereo part and the T265 for the Imu. I can send you my yaml used for the two cameras if you want

antithing, 2021-04-10:
I was thinking about using the T265 for stereo and imu, what is the reason that you used both cameras? Is the fisheye less accurate?

ryleymcc, 2021-04-10:
yea Im trying to get the t265 to work in stereo_inertial but i have the d435i too so send your yamls

Shkunn, 2021-04-10:
> I was thinking about using the T265 for stereo and imu, what is the reason that you used both cameras? Is the fisheye less accurate?

I don't know if the fisheye is less accurate but I had both the D435 and T265 with me so I decided to use them together as suggested by Intel on their website. 

But maybe the T265 only could work !

Shkunn, 2021-04-10:
> Yea Im trying to get the t265 to work in stereo_inertial but i have the d435i too so send your yamls. Everything helps.

Here is the yaml i use for the D435 : 

%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 392.008850097656
Camera.fy: 392.008850097656
Camera.cx: 321.213836669922
Camera.cy: 238.390502929688

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

Camera.bFishEye: 0

Camera.width: 640
Camera.height: 480

# Camera frames per second 
Camera.fps: 30.0

# stereo baseline times fx
Camera.bf: 49.974

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 35.0

# #--------------------------------------------------------------------------------------------
# # Stereo Rectification. Only if you need to pre-rectify the images.
# # Camera.fx, .fy, etc must be the same as in LEFT.P
# #--------------------------------------------------------------------------------------------
# LEFT.height: 720
# LEFT.width: 1280
# LEFT.D: !!opencv-matrix
#    rows: 1
#    cols: 5
#    dt: d
#    data:[-0.00604297406971455, 0.0451732017099857, -0.0425471812486649, 0.00776465190574527, 0]
# LEFT.K: !!opencv-matrix
#    rows: 3
#    cols: 3
#    dt: ds
#    data: [286.3045, 0.0, 417.1953, 0.0, 286.2737, 399.1153, 0.0, 0.0, 1.0]
# LEFT.R:  !!opencv-matrix
#    rows: 3
#    cols: 3
#    dt: d
#    data: [0.999966347530033, -0.001422739138722922, 0.008079580483432283, 0.001365741834644127, 0.9999741760894847, 0.007055629199258132, -0.008089410156878961, -0.007044357138835809, 0.9999424675829176]
# LEFT.P:  !!opencv-matrix
#    rows: 3
#    cols: 4
#    dt: d
#    data: [653.348083496094, 0, 642.023071289062, 0,  0, 653.348083496094, 357.317504882812, 0,  0, 0, 1, 0]

# RIGHT.height: 720
# RIGHT.width: 1280
# RIGHT.D: !!opencv-matrix
#    rows: 1
#    cols: 5
#    dt: d
#    data:[-0.00477181980386376, 0.0391876511275768, -0.0359093882143497, 0.00556322885677218, 0]
# RIGHT.K: !!opencv-matrix
#    rows: 3
#    cols: 3
#    dt: d
#    data: [286.6798, 0.0, 422.400, 0.0, 286.5119, 398.7724, 0.0, 0.0, 1]
# RIGHT.R:  !!opencv-matrix
#    rows: 3
#    cols: 3
#    dt: d
#    data: [0.9999633526194376, -0.003625811871560086, 0.007755443660172947, 0.003680398547259526, 0.9999684752771629, -0.007035845251224894, -0.007729688520722713, 0.007064130529506649, 0.999945173484644]
# RIGHT.P:  !!opencv-matrix
#    rows: 3
#    cols: 4
#    dt: d
#    data: [653.348083496094, 0, 642.023071289062, 0,  0, 653.348083496094, 357.317504882812, 0,  0, 0, 1, 0]

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 600

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.8

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 12

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

ryleymcc, 2021-04-14:
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/297

aravindSolteq, 2021-04-23:
> Yes of course ! This is the one I use for Stereo only with the T265 :
>
> %YAML:1.0
>
> #--------------------------------------------------------------------------------------------
> # Camera Parameters. [...]

How did you get this calibration parameters?? Is it from intel real sense or you calibraed using some tool? 
Also How did you give only one camera calibration values when there are two(left and right camera)
Your help is really appreciated.

Thanks!

jdtardos, 2021-12-23:
Version v1.0 includes a calibration tutorial and examples to run ORB-SLAM3 with Intel Cameras.

closed 2021-12-23 by jdtardos