UZ-SLAMLab/ORB_SLAM3 #285: Issues with map resetting, freezing and random motions

MrOCW, 2021-04-02:
Hi everyone, I am using a Intel Realsense D435i and currently attempting to run it on a Jetson Xavier NX.

After some digging around, this is my yaml file:
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 919.915
Camera.fy: 920.177
Camera.cx: 643.488
Camera.cy: 361.480

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

Camera.width: 1280
Camera.height: 720

# Camera frames per second 
Camera.fps: 30

# IR projector baseline times fx (aprox.)
Camera.bf: 13.844

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 40.0

# Deptmap values factor
DepthMapFactor: 1.0

#--------------------------------------------------------------------------------------------
# Modeling Realsense D435 Sensor Noise for Improved 3D Reconstruction and Tracking.
sensor_processor/type: structured_light
sensor_processor/cutoff_min_depth: 0.2
sensor_processor/cutoff_max_depth: 3.25
sensor_processor/normal_factor_a: 0.000611
sensor_processor/normal_factor_b: 0.003587
sensor_processor/normal_factor_c: 0.3515
sensor_processor/normal_factor_d: 0
sensor_processor/normal_factor_e: 1
sensor_processor/lateral_factor: 0.01576

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

1) Is Pinhole the only option for CameraType? What should it be for a D435i?

2) When running ORB SLAM3, the Graph and Keyframes are strangely, all over the place,
The map also constantly resets as I move the camera, erasing all the points. After awhile, everything freezes and the cmd line shows 
id last: 222    id curr: 223
id last: 222    id curr: 224
id last: 222    id curr: 225
id last: 222    id curr: 226
................ as id curr keeps increasing
[image]

Have been trying to fix these issues for so long and its really frustrasting. Any help is greatly appreciated!

Sherlock-hh, 2021-11-04:
hello，I met the same problem , have you fix it yet, I dig with code, It's shows the last frame 'timestamps didn't refresh, so the diff between current frame is keeping increasing, so every time it's come to this code ,it throw the info, should I just reset the map? looking for your reply, thank you very much

Maithilishetty, 2022-03-22:
@MrOCW have you fixed this yet?

hseysen, 2023-09-06:
I suspect the issue may be due to the calibration, which results in inaccurate feature tracking, ultimately losing track of the features and moving randomly.

> Camera.k1: 0.0
> Camera.k2: 0.0
> Camera.p1: 0.0
> Camera.p2: 0.0

These values in the `.yaml` file lead me to believe that you have not calibrated the camera, or may have forgotten to add the distortion coefficients. See [Kalibr](https://github.com/ethz-asl/kalibr/wiki/camera-imu-calibration) for Camera-IMU calibration.