UZ-SLAMLab/ORB_SLAM3 #286: Segmentation fault (core dumped)

ryleymcc, 2021-04-05:
Using this command works perfectly.
./Examples/Stereo-Inertial/stereo_inertial_euroc ./Vocabulary/ORBvoc.txt ./Examples/Stereo-Inertial/EuRoC.yaml ~/Datasets/EuRoc/MH01 ./Examples/Stereo-Inertial/EuRoC_TimeStamps/MH01.txt dataset-MH01_stereoi

When I run:
rosrun ORB_SLAM3 RGBD /home/ryley/Dev/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/ryley/Dev/ORB_SLAM3/Examples/RGB-D/d435i.yaml
I get the error:

```
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D
Segmentation fault (core dumped)
```

ryleymcc, 2021-04-05:
can someone explain how would make a d435i.yaml file properly?

sumitsarkar1, 2021-04-07:
make a fresh install of ubuntu if its an option..or purge remove all opencv and install opencv 3.4.6. If u want to experiment with different opencv versions then change the install directory while building opencv (if you are building from source)

ryleymcc, 2021-04-07:
I have purged opencv from my ROS and all other instances. I am currently re-installing opencv 4.5.1 and going off [this](https://github.com/shanpenghui/ORB_SLAM3_Fixed) repo I also have to re install opencv 4.2 for ROS dependencies using the command sudo apt-get install linopencv-dev.

ryleymcc, 2021-04-07:
re installed ros-noetic-opencv-bridge. Still getting the warnings after ./build_ros.sh:
`[ 92%] Linking CXX executable ../Mono
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_core.so.4.5
[ 92%] Built target Mono
[100%] Linking CXX executable ../MonoAR
/usr/bin/ld: warning: libopencv_imgproc.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_imgproc.so.4.5
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_core.so.4.5
[100%] Built target MonoAR

`

sumitsarkar1, 2021-04-07:
your ROS opencv version is different from the version you used to build ORB_SLAM3. It comes as a warning while you run ./build_ros.sh but causes error while running TUM and Euroc examples

closed 2021-04-07 by ryleymcc

ryleymcc, 2021-10-31:
> hello! I met the same problem, have you solved it? dose reinstall the opencv to the opencv3 can help? looking for your reply, thank you very much!

There is another repo that fixes the issue. I put a link to it in my comment from a few months ago.

terrense, 2022-10-18:
I am using OpenCV 4.5.5 , also met this problem, please help