UZ-SLAMLab/ORB_SLAM3 #287: ROS topics for output

stephenmcc, 2021-04-06:
There don't appear to be any ROS output topics for the estimated pose or map, at least for the default settings. It'd be helpful if those could be added in.

pinkfloydfan, 2021-04-06:
Hi,

I've partially solved this myself. Kindly refer to my fork of ORB SLAM3 [here](https://github.com/pinkfloydfan/ORB_SLAM3/blob/83d41a1e924c6b9e45ce0e2aca7398107c3bb111/Examples/ROS/ORB_SLAM3/src/ros_mono_inertial.cc). I've only done it for the inertial node, however I imagine it'll be similar for the other nodes.

The `ImageGrabber` class will have a method called to `mpSLAM->TrackMonocular` which will return a transformation matrix. By default the return variable is not stored, modify `ImageGrabber` to contain a `ros::Publisher` and use it to publish pose. It seems older versions of ORB SLAM2 also had the same issue, and I followed the gist of [this article](https://roboticsknowledgebase.com/wiki/state-estimation/orb-slam2-setup/) to make the appropriate modifications. One thing you have to do is change `ORB_SLAM2::Converter::toQuaternion` to `ORB_SLAM3::Converter::toQuaternion`.

Hope this helps.

stephenmcc, 2021-04-27:
@pinkfloydfan Appreciate the info, I was able to publish a ROS transform from that `TrackMonocular` method. However I'm seeing that the axes are: +x when camera moves left, +y when camera moves up, and +z when camera moves backward. This seems odd since this appears to be a left handed coordinate frame, and that +z usually means forward motion.The pangolin window is showing that the SLAM is doing a good job though, so it isn't estimating incorrectly. Any thoughts?

pinkfloydfan, 2021-04-28:
@stephenmcc In the pose estimate I'm getting, +z is forwards, -z is backwards, -y is up, +y is down, -x is left, +x is right. Are you using mono-inertial? It might be worth checking the Tbc matrix in the configuration yaml file, which is the transfrom from body coordinates to camera coordinates. Did you use Kalibr or a similar tool?

stephenmcc, 2021-04-28:
I just needed to invert the transform that TrackMonocular returned, which I realized after checking your fork. Really appreciate the help!

titanX108, 2023-06-13:
@pinkfloydfan Thank you for your code. I have a small issue. How did you get the  output from mpSLAM->TrackMonocular. I'm getting this error.

/Examples_old/ROS/ORB_SLAM3/src/ros_mono_inertial.cc:179:50: error: no match for 'operator=' (operand types are 'cv::Mat' and 'Sophus::SE3f' {aka 'Sophus::SE3<float>'})
  179 |       T_ = mpSLAM->TrackMonocular(im,tIm,vImuMeas);
      |

Rajni-kanth, 2023-11-07:
@stephenmcc , would you give us some clue how we can do that , I am using orbslam3 but my robot was moving on upward direction