UZ-SLAMLab/ORB_SLAM3 #289: Segmentation fault in SetNewBias(...)

laxnpander, 2021-04-08:
Hey,

I am running ORB SLAM3 with a custom setup (from rosbag). The initialization is successful, though I believe something is going on with the IMU. After a few frames it is crashing with a segmentation fault. Here is the backtrace:

First KF:0; Map init KF:0
New Map created with 480 points
Point distribution in KeyFrame: left-> 480 --- right-> 0
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
scale too small
build optimization graph
Not preintegrated measurement

Thread 24 "Mono_Inertial" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff58ffd700 (LWP 29867)]
__GI___pthread_mutex_lock (mutex=0x5f0) at ../nptl/pthread_mutex_lock.c:65
65	../nptl/pthread_mutex_lock.c: No such file or directory.
(gdb) bt
#0  0x00007ffff782afa0 in __GI___pthread_mutex_lock (mutex=0x5f0) at ../nptl/pthread_mutex_lock.c:65
#1  0x00007ffff6aa7b8b in ORB_SLAM3::IMU::Preintegrated::SetNewBias(ORB_SLAM3::IMU::Bias const&) () at /media/hdd/Workspace/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#2  0x00007ffff69d8444 in ORB_SLAM3::Optimizer::InertialOptimization(ORB_SLAM3::Map*, Eigen::Matrix<double, 3, 3, 0, 3, 3>&, double&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&, bool, Eigen::Matrix<double, -1, -1, 0, -1, -1>&, bool, bool, float, float) () at /media/hdd/Workspace/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#3  0x00007ffff6939506 in ORB_SLAM3::LocalMapping::InitializeIMU(float, float, bool) () at /media/hdd/Workspace/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#4  0x00007ffff693b3b8 in ORB_SLAM3::LocalMapping::Run() () at /media/hdd/Workspace/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#5  0x00007ffff487cd80 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff78286db in start_thread (arg=0x7fff58ffd700) at pthread_create.c:463
#7  0x00007ffff42c4a3f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Any guesses?

herohuyongtao, 2021-04-12:
This seems to be fixed by modifing `Optimizer.cc` as follows, but not sure their effects to the result.

[image]

laxnpander, 2021-04-20:
@herohuyongtao Thanks for your suggestion. For me it was related to IMU - Camera misalignment. So I fixed this and the error didnt occure anymore.

closed 2021-04-20 by laxnpander

herohuyongtao, 2021-04-20:
> @herohuyongtao Thanks for your suggestion. For me it was related to IMU - Camera misalignment. So I fixed this and the error didnt occure anymore.

@laxnpander Glad to hear this. Is the misalignment of timestamp or coordinate system?

panping1314, 2022-01-20:
@laxnpander Hello, how did you modify it? I also have this problem.

panping1314, 2022-01-20:
@laxnpander Hello, how did you modify it? I also have this problem

laxnpander, 2022-01-20:
@panping1314 Cant remember exactly how I solved it. But I think it was a bad geometric calibration of my custom system using KalibR. So make sure the alignment between camera and imu frame is correct.