UZ-SLAMLab/ORB_SLAM3 #296: Running EuRoC error on Ros melodic  (Aborted (core dumped))

Mingchi-Yeh, 2021-04-13:
Hi, I'm trying to run ORB_SLAM3 on Ros melodic , the RGBD and Mono are working fine,but when I try to use Stereo_Inertial,Mono_Inertial or Stereo,I got the same error as following :

`rosrun ORB_SLAM3 Stereo_Inertial /home/mingchi/catkin_ws/src/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/mingchi/catkin_ws/src/ORB_SLAM3/Examples/Stereo-Inertial/EuRoC.yaml true

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 435.205
- fy: 435.205
- cx: 367.452
- cy: 252.201
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 20
- color order: RGB (ignored if grayscale)

Depth Threshold (Close/Far Points): 3.85272

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
double free or corruption (out)
Aborted (core dumped)`

I have no idea what reason that it "Aborted (core dumped)".
Could you give any suggestion to solve this problem?

sumitsarkar1, 2021-04-13:
@Mingchi-Yeh  please mention the rosbag you are playing

Mingchi-Yeh, 2021-04-14:
I playing EuRoC dataset (MH_01_easy) and rosrun the Stereo_Inertial.

rosbag play --pause MH_01_easy.bag /cam0/image_raw:=/camera/left/image_raw /cam1/image_raw:=/camera/right/image_raw /imu0:=/imu

rosrun ORB_SLAM3 Stereo_Inertial /home/mingchi/catkin_ws/src/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/mingchi/catkin_ws/src/ORB_SLAM3/Examples/Stereo-Inertial/EuRoC.yaml true

ryleymcc, 2021-04-16:
Can you rosrun with gdb and execute a backtrace and post the output.

Mingchi-Yeh, 2021-04-28:
I run gdb and find the program exit abnormally on "ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::IMU_MONOCULAR,true)",and terminal show the "double free or corruption (out)
Aborted (core dumped)".
What should I do?

ahuzsg, 2021-09-23:
same problem with me ! have you solved?