UZ-SLAMLab/ORB_SLAM3 #297: t265 .yaml settings

ryleymcc, 2021-04-13:


sumitsarkar1, 2021-04-16:
I am facing same problem with Stereo and a TUM yaml file. All I can think of is convert T265.yaml (presently looks like a TUM format yaml) in EuRoC.yaml format and check. This is because all Stereo and Stereo-Inertial for EuRoC data sets are running fine for me. Stereo for TUM is not working AT ALL...Stereo-Inertial for TUM is working but not all cases.

aravindSolteq, 2021-05-03:
Do we need rectification of Fisheye images? Because their instruction says that we may not need to rectify the images for fisheye lens

sumitsarkar1, 2021-05-05:
Can you point where it says that rectification not required. I guess we have to rectify...thats why it requires the yaml file.

aravindSolteq, 2021-05-05:
https://github.com/UZ-SLAMLab/ORB_SLAM3#running-stereo-node

sumitsarkar1, 2021-05-05:
What you have shared in the launch file is the TUM format where `Camera.type: "KannalaBrandt8"` . Some how the TUM Stereo Examples are not working even for TUM bench mark data sets . So the way round is ...provide the t265 settings in EuRoC format (EuRoC.yaml in `/Examples/Stereo` ). And if you notice in the EuRoC.yaml format `Camera.type: "PinHole"` for which images should be rectified. Hence yaml file is required for PinHole camera type or should be pre rectified if you set `ONLINE_RECTIFICATION` as `False`

aravindSolteq, 2021-05-05:
But for EuRoC.yaml has this Calibration and distortion parameters for only one camera and it says ("equal for both cameras after stereo rectification"). How to do this stereo rectification?

Also for pre-rectify were we need to give the rectification matrices, How to get the matrices P ? Is there any tool to obtain this directly. I used kalibr to get the Distortion coefficients and K matrices.

sumitsarkar1, 2021-05-05:
1.) I have not used Kalibr but is good....Realsense itself provides these K and P matrices....or you can use opencv... I use the calibration application provided with [Rtabmap](https://github.com/introlab/rtabmap) (which in turn uses OpenCV at the backend)

2.)  Stereo Rectification is the joint rectification of the pair of images to produce the camera parameters. Its there in OpenCV [here](https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga617b1685d4059c6040827800e72ad2b6)

3.)  I (long time back) used [this](https://github.com/ethz-asl/image_undistort) to rectify individual T265 images and then feed it to the algorithm as two pin hole cams. Or you can use this from [ROS](http://wiki.ros.org/stereo_image_proc)

aravindSolteq, 2021-05-17:
@sumitsarkar1 : I have calibrated the camera with rtabmap. But the calibration is done using "equidistant" model(I suppose this isnt KannalaBrandt8 model). But can you share me your yaml file that you have used for ORB SLAM 3. Especially if you have tried the "Stereo-Inertial" mode. My map gets resets in between mapping when there is a movement of camera. Have you faced this kind of issue

sumitsarkar1, 2021-05-17:
I have faced all the same issues which you have mentioned till now..I am able to only run Stereo_Inertial on EuRoC data set ...even simple Mono mode on benchmark data is not functional ...i.e. after initial struggle of map initialisation the track abruptly stops. These were not the case with ORB_SLAM2. With ORB_SLAM2 I wrote a ROS node with OpenCV to rectify the T265 single camera (treating the T265 as mono fisheye ) then feed it to ORB_SLAM2 node and it worked fine. I hope the owners come up with some solution. Till then I found an alternative [here](https://github.com/OpenVSLAM-Community/openvslam/tree/feat/imu). It allows map saving too unlike ORB_SLAM. You can try .

jdtardos, 2021-12-23:
Version v1.0 now includes a calibration tutorial and examples with camera T265

closed 2021-12-23 by jdtardos