UZ-SLAMLab/ORB_SLAM3 #300: compile error with no match operator

kiddjacky, 2021-04-23:
Hi,

I have the following compile error. Is that I missing any lib or dependency? Thanks

/mnt/c/Users/Kidd/github/ORB_SLAM3/src/LocalMapping.cc:628:49: error: no match for ‘operator/’ (operand types are ‘cv::Matx<float, 3, 1>’ and ‘float’)
                 x3D = x3D_h.get_minor<3,1>(0,0) / x3D_h(3);
                       ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~

PattyLiang, 2021-04-23:
hi,

I have the same problem, and I use below function to workaround.

`x3D = cv::Matx31f(x3D_h.get_minor<3,1>(0,0)(0) / x3D_h(3), x3D_h.get_minor<3,1>(0,0)(1) / x3D_h(3), x3D_h.get_minor<3,1>(0,0)(2) / x3D_h(3));`

aravindSolteq, 2021-04-23:
This error occurs in multiple places

aravindSolteq, 2021-04-23:
Solution by @PattyLiang is solving the issue. But also replace it in ORB_SLAM3/src/CameraModels/KannalaBrandt8.cpp

piphagor18, 2021-04-23:
> Solution by @PattyLiang is solving the issue. But also replace it in ORB_SLAM3/src/CameraModels/KannalaBrandt8.cpp

You can add below code to KannalaBrandt8.h.

namespace cv
{
    template<typename _Tp, int m, int n> static inline
        Matx<_Tp, m, n> operator / (const Matx<_Tp, m, n>& a, float alpha)
    {
        return Matx<_Tp, m, n>(a, 1.f / alpha, Matx_ScaleOp());
    }
}

It will works for all same problems

silencht, 2021-04-26:
I also have the same problem, and by @piphagor18 is solving my errors(: no match for ‘operator/’ ) on LocalMapping.cc and KannalaBrandt8.cpp files. 
But, I also have the problem again: /usr/bin/ld: not found -lboost_serialization………… T^T

kiddjacky, 2021-04-26:
@PattyLiang  and @piphagor18  Thanks for the solution. It works for me.

jjgr3496, 2021-04-30:
Hi everybody,

This is an issue related with the OpenCV version (you are likely using OpenCV 4.1 or 4.2). In those versions, the operator ‘operator/’ (operand types are ‘cv::Matx<float, 3, 1>’ and ‘float’) is not implemented for some reason. We recommend you upgrading your OpenCV version to, at least, 4.4.

Best regards!

closed 2021-04-30 by jjgr3496

Ahmad-Drak, 2021-04-30:
> I also have the same problem, and by @piphagor18 is solving my errors(: no match for ‘operator/’ ) on LocalMapping.cc and KannalaBrandt8.cpp files.
> But, I also have the problem again: /usr/bin/ld: cannot find-lboost_serialization………… T^T

for that you need the boost library. sudo apt install libboost

silencht, 2021-05-10:
Thanks~ 
I  use below steps to solve this problem:

method1: 

```
#安装locate库
sudo apt-get install locate
#定位缺失库位置
locate libboost_serialization.so
#若可定位到（如定位结果为/usr/local/ahuasdk/libXXX.so）进行软链接，定位不到见方法二
sudo ln -s /usr/local/ahuasdk/libXXX.so /usr/lib/libXXX.so
```
method2:

```
sudo apt-get install apt-file
sudo apt-file update
apt-file search libboost_serialization.so
#apt-file 将列出所有包含libXXX.so文件的apt包，
#选择相应的包用apt-get安装即可，通常请选择带dev的包安装。输出举例如下
libboost_serializatio1.65-dev: /usr/lib/aarch64-linux-gnu/libboost_serialization.so
#安装缺失库
sudo apt-get install libboost-serializatio1.65-dev
#注意，可能为
sudo apt-get install libboost-serializatio1.65-dev
```
or method3?

```
sudo apt-get install libboost-dev
```