UZ-SLAMLab/ORB_SLAM3 #301: Save/Load Atlas V0.4

neo-knight-td, 2021-04-26:
Was the save/load atlas functionality implemented in version V0.4 ? It seems like a lot of code related to serialization and atlas save/load has been deleted in that version. Is it the point to release it later ?

fliegend, 2021-05-06:
Same concern. Actually save/load atlas could work with minor modifications in serialize(Archive &ar, const unsigned int version). Hope it will be released later

neo-knight-td, 2021-05-06:
@fliegend meanwhile I made a pull request with the Save/Load Functionnality. You can review it here https://github.com/UZ-SLAMLab/ORB_SLAM3/pull/310. Hope it helps

fliegend, 2021-05-06:
> @fliegend meanwhile I made a pull request with the Save/Load Functionnality. You can review it here #310. Hope it helps

Many thanks:). Another problem from my side is that it saves too much data, which makes the stored .bin file too large. Any ideas to shrink saved map size?

neo-knight-td, 2021-05-06:
@fliegend You're welcome ! 
On what device are you running ORB SLAM 3? Do you come to a shortage in memory on your device ? Or is it the serialization that gives you an error? 
Serialization is just transforming the classes in a bit stream. So if you perform compression on that stream you should be able to reduce the size of your file. This should be useful I think https://stackoverflow.com/a/48034742/15539525. It is part of boost so you should not have to perform that many changes to the code.

fliegend, 2021-05-07:
> @fliegend You're welcome !
> On what device are you running ORB SLAM 3? Do you come to a shortage in memory on your device ? Or is it the serialization that gives you an error?
> Serialization is just [...]

Probably this is a good solution, I'll have a try, thanks!

treememory, 2021-06-12:
do not save image(mat), cloud shrink lot

Jiaxin8122, 2021-08-21:
> > @fliegend You're welcome !
> > On what device are you running ORB SLAM 3? Do you come to a shortage in memory on your device ? Or is it the serialization that gives you an error?
> > Serialization is just transforming the classes in a bit stream. So if you perform compression on that stream you should be able to reduce the size of your file. This should be useful I think https://stackoverflow.com/a/48034742/15539525. It is part of boost so you should not have to perform that many changes to the code.
> 
> I run in ubuntu 16.04 with ros kinetic. Memory isn't a problem, just the saved map is larger compared to ORB-SLAM2.
> Probably this is a good solution, I'll have a try, thanks!

Hello, 

I can't find the V0.4 version but the newest version has no function like Save/Load . I use ubuntu 16.04 and ros as well. Could you share the V0.4 version? Maybe we can add wechat or with email.  austin.jxhuang@gmail.com

regards

jdtardos, 2021-12-23:
Version v1.0 includes map loading and saving operations

closed 2021-12-23 by jdtardos