UZ-SLAMLab/ORB_SLAM3 #302: Monocular (without inertial) examples fail with 0.4

Bloodwyn, 2021-04-26:
Hi, it seems like the Monocular-only examples don't work with the newest version. 
Everything seems to get loaded in properly, but the system won't track anything.
With any TUM dataset it will only play the video but does not show any keypoints nor the "trying to intialize" message.

As Monocular-intertial still works in 0.4 and Monocular-only used to work in previous version I expect a bug in the latest release.

Can someone confirm this? Thanks.

Bloodwyn, 2021-04-26:
FIX: change line 106 in file mono_tum_vi.cc to use cv::IMREAD_GRAYSCALE

closed 2021-04-26 by Bloodwyn

sadsasuke, 2022-03-17:
Hi, I am facing the same issue. I changed the line 106 but still same problem. any help . @Bloodwyn ?

sadsasuke, 2022-03-17:
> FIX: change line 106 in file mono_tum_vi.cc to use cv::IMREAD_GRAYSCALE

Can you please tell me the steps you did after you made the change in .cc file .. I am a bit new to coding. Thanks in advance