UZ-SLAMLab/ORB_SLAM3 #303: Is using cv::undistortPoints function in the Frame:UndistortKeyPoints() justified for fisheye camera?

srippa-nexar, 2021-04-27:
Hi all,

The method `Frame:UndistortKeyPoints()` uses the OpenCV function `cv::undistortPoints` to undistort points even when the camera model is `KannalaBrandt8` (OpenCV Fisheye model). This method is called during the constructor of `Frame` and sets the member variable `mvKeysUn` containing the undistorted points. As far as I understand those are the wrong undistorted points when the camera model is of type `KannalaBrandt8`.

In the tracking module (`Tracking::MonocularInitialization`, line 2090) those undistorted points are sent to `mpCamera->ReconstructWithTwoViews` which in case of a `KannalaBrandt8` camera (file `KannalaBrandt8.cpp`, line 219) gets undistorted again, this time using the proper `cv::fisheye::undistortPoints`, resulting in another distortion of the points that were already udistorted by `Frame:UndistortKeyPoints`

My question is: Am I missing something or this is a real issue?

srippa-nexar, 2021-04-29:
I managed to understand the code. It seems to be correct: When we have a pinhole camera, everything work as expected and keypoints get properly undistorted and stored in `mvKeysUn`. The `ReconstructWithTwoViews` code for the pinhole derived class than work on the undiosordet points and all is good.

If the camera however is of type `KannalaBrandt8` then in the `Frame` class we set the vector of distortion coefficients to zero which cause the  `Frame:UndistortKeyPoints` to do nothing so the vector `mvKeysUn` is populated with the **distorted** vector of keypoints. When the `ReconstructWithTwoViews` method of the `KannalaBrandt8` camera is called with these undistorted point, is undistort them using the proper `cv::fisheye::undistortPoints` function which does the right thing.

This is correct but confusing. IMHO it would be better adding a virtual method `UndistortPoints` to `GeometricCamera.h` which will be properly implemented by the derived classes. Then in `Frame:UndistortKeyPoints()` this function would be called to undistort the points and always return undistorted points in `mvKeysUn`.

amberwood31, 2021-08-25:
The issue with the proposed way is that we don't always want the undistorted points in `mvKeysUn`. 

When `KannalaBrandt8` camera is used, its `projectJac` considers the distorted pixel coordinates. Accordingly, all the measurements set in the optimizer should be the distorted pixel coordinates. Since `mvKeysUn` is used inside the `PoseOptimization` function to set the measurements, which are used in the tracking thread. The variable should contain the distorted pixel coordinates when `KanalaBrandt8` camera model is used.

The name `mvKeysUn` might give user a wrong idea. But it does make the switching of `Pinhole` and `KannalaBrandt8` camera model rather simple in the current implementation.

soeroesg, 2021-08-25:
a related comment: the Pinhole.h camera model does not store the distortion parameters, and none of the camera models store the original image width and height, but these could be useful later down in the pipeline.