UZ-SLAMLab/ORB_SLAM3 #305: t265 + Stereo_Inertial + Ros + realtime

aravindSolteq, 2021-04-30:
Hi, 
       I'm currently trying to run ORB SLAM 3 with Intel realsense t265 in Stereo Inertial mode. I can run the ORB SLAM 3, but the map resets for every 10 frames and I couldnt solve it by any means 
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 520
mnInitialFrameId = 518
431 Frames set to lost
not IMU meas
First KF:105; Map init KF:104
New Map created with 245 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 523
mnInitialFrameId = 521
433 Frames set to lost
not IMU meas
First KF:106; Map init KF:105
New Map created with 214 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 526
mnInitialFrameId = 524
435 Frames set to lost
not IMU meas
First KF:107; Map init KF:106
New Map created with 246 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...

can you please help me wtih this solution

sumitsarkar1, 2021-04-30:
share your yaml file please

aravindSolteq, 2021-05-03:
Here is my yaml file for t265.  
```
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Left Camera calibration and distortion parameters (OpenCV)
Camera.fx: 306.56357097653495 
Camera.fy: 305.56042118353184
Camera.cx: 434.2944063512854
Camera.cy: 394.2731238071397

# PinHole distortion parameters
Camera.k1: -0.28394309239916604
Camera.k2: 0.06846916170247387
Camera.p1: 6.813344248087167e-05
Camera.p2: -0.003363583581458696

# Right Camera calibration and distortion parameters (OpenCV)
Camera2.fx: 307.0145428737875 
Camera2.fy: 305.70433211159366
Camera2.cx: 434.39538239909575
Camera2.cy:  392.8478563244833

# PinHole distortion parameters
Camera2.k1: -0.28770827311876207 
Camera2.k2: 0.07146524167094274
Camera2.p1: 0.00045536748109210147
Camera2.p2: -0.0023976845889413247

# Transformation matrix from right camera to left camera
Tlr: !!opencv-matrix
  rows: 3
  cols: 4
  dt: f
  data: [ 0.9999411252356276, 0.001724774265428219, 0.010713132886021406, -0.06357291767834125,
          -0.0017355224147441234, 0.999997999908411, 0.000994052875333053, 6.242827075813476e-05,
          -0.010711396941956615, -0.0010125872329600702, 0.9999421186461949, -0.0007602738114222626]

# Lapping area between images
Camera.lappingBegin: 0
Camera.lappingEnd: 847

Camera2.lappingBegin: 0
Camera2.lappingEnd: 847

# Camera resolution
Camera.width: 848
Camera.height: 800

# Camera frames per second 
Camera.fps: 30.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 35.0
Camera.bf: 19.490543672

# Transformation from body-frame (imu) to left camera
Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [-0.99967008, 0.00383731, -0.02539712, 0.00381778,
          -0.00379978, -0.99999162, -0.00152601,  0.0547511,
         -0.02540276, -0.00142901, 0.99967628, -0.03778417,
          0.0, 0.0, 0.0, 1.0]

# IMU noise (Use those from VINS-mono)
IMU.NoiseGyro: 0.00531091888368 # 0.004 (VINS) # 0.00016 (TUM) # 0.00016    # rad/s^0.5 
IMU.NoiseAcc: 0.0188670828051 # 0.04 (VINS) # 0.0028 (TUM) # 0.0028     # m/s^1.5
IMU.GyroWalk: 2.01765706199e-05 # 0.000022 (VINS and TUM) rad/s^1.5
IMU.AccWalk: 0.000515451463717 # 0.0004 (VINS) # 0.00086 # 0.00086    # m/s^2.5
IMU.Frequency: 200

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000 # Tested with 1250

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20 # 20
ORBextractor.minThFAST: 7 # 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -3.5
Viewer.ViewpointF: 500
```

iamchoking, 2021-07-12:
It looks like your camera type is set to "PinHole" this is not encouraged since the camera is actually fish eye.see #88 for details.

Chenliezhen259, 2021-12-08:
However，camera type is to the same type with real camera，with Stereo_Inertial + Ros + realtime，it also the map resets for every 10 frames and it could not mapping.