UZ-SLAMLab/ORB_SLAM3 #308: ORB_SLAM3 and g2o given won't in jetson Xavier NX with Opencv 4.1.1

FPSychotic, 2021-05-03:
Hi using the script g2o won't compile. II got compile the g2o version that came with orb-slam2 .
Using it, I start compile orb-slam3 but It send this error at the end ( but also others that get lost in the terminal history)
I use a Jetson Xavier NX, with Opencv 4.1.1 compiled with cuda and cudnn.
```
[ 69%] Linking CXX shared library ../../../Thirdparty/g2o/lib/libg2o.so
[ 69%] Built target g2o
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc: In member function ‘bool ORB_SLAM3::Tracking::ParseORBParamFile(cv::FileStorage&)’:
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:1101:100: warning: ‘nLevels’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         mpIniORBextractor = new ORBextractor(5*nFeatures,fScaleFactor,nLevels,fIniThFAST,fMinThFAST);
                                                                                                    ^
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:1101:100: warning: ‘fScaleFactor’ may be used uninitialized in this function [-Wmaybe-uninitialized]
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:1104:41: warning: ‘nFeatures’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     cout << "- Number of Features: " << nFeatures << endl;
                                         ^~~~~~~~~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc: In member function ‘bool ORB_SLAM3::Tracking::ParseCamParamFile(cv::FileStorage&)’:
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:641:27: warning: ‘k4’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float k1, k2, k3, k4;
                           ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:641:23: warning: ‘k3’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float k1, k2, k3, k4;
                       ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:641:19: warning: ‘k2’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float k1, k2, k3, k4;
                   ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:641:15: warning: ‘k1’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float k1, k2, k3, k4;
               ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:640:27: warning: ‘cy’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                           ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:640:23: warning: ‘cx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                       ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:640:19: warning: ‘fy’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                   ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:640:15: warning: ‘fx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
               ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:505:27: warning: ‘cy’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                           ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:505:23: warning: ‘cx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                       ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:505:19: warning: ‘fy’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
                   ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:505:15: warning: ‘fx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         float fx, fy, cx, cy;
               ^~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc: In member function ‘bool ORB_SLAM3::Tracking::ParseIMUParamFile(cv::FileStorage&)’:
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:1208:52: warning: ‘Ngw’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     mpImuCalib = new IMU::Calib(Tbc,Ng*sf,Na*sf,Ngw/sf,Naw/sf);
                                                 ~~~^~~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:1208:45: warning: ‘Na’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     mpImuCalib = new IMU::Calib(Tbc,Ng*sf,Na*sf,Ngw/sf,Naw/sf);
                                           ~~^~~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:1208:39: warning: ‘Ng’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     mpImuCalib = new IMU::Calib(Tbc,Ng*sf,Na*sf,Ngw/sf,Naw/sf);
                                     ~~^~~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:1138:11: warning: ‘freq’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     float freq, Ng, Na, Ngw, Naw;
           ^~~~
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc: In member function ‘void ORB_SLAM3::Tracking::Track()’:
/home/nx/orbslam3/ORB_SLAM3/src/Tracking.cc:1980:13: warning: ‘bOK’ may be used uninitialized in this function [-Wmaybe-uninitialized]
             if(bOK)
             ^~
CMakeFiles/ORB_SLAM3.dir/build.make:250: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
CMakeFiles/Makefile2:300: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
make: *** [all] Error 2

```

AnhTuDo1998, 2021-05-04:
Hi @FPSychotic, not sure this is applicable to you but no harm trying. Anw, I was using ORB-SLAM2 and has the same problem (on Jetson Xavier NX). My solution is to actually remove the OpenCV 4.1.1 comes with the Jetpack and install OpenCV 3.2.2. You may refer to the following gist. Hope it helps

https://gist.github.com/AnhTuDo1998/baf50e5496d27b15b7d89767c4e45c6e

FPSychotic, 2021-05-04:
> Hi @FPSychotic, not sure this is applicable to you but no harm trying. Anw, I was using ORB-SLAM2 and has the same problem (on Jetson Xavier NX). My solution is to actually remove the OpenCV 4.1.1 [...]

I was thinking in go to 3.4.x I would try 3.4.14 for netter support for cuda (I read or unterstood 3.4.6 and newer suport better cuda).
The problem is I dont know if it will affect to deepstream,tensorrt and other nvidia stuff.
had you any  isue with downgrade?

AnhTuDo1998, 2021-05-04:
Hi, did not try Deepstream and CUDA so not sure but TensorRT seems to works fine.