UZ-SLAMLab/ORB_SLAM3 #311: Mono_Inertial on ROS Consistent Segmentation Fault

rdehaggart1, 2021-05-05:
I am trying to run the [MidAir](https://midair.ulg.ac.be/index.html) dataset on ORB-SLAM3 Mono_Inertial. 
I haven't seen it used on SLAM before, but I've managed to get it working great for ORB-SLAM2 (and ORB-SLAM3 Monocular). However, every time I try to run it on ORB-SLAM3 Mono_Inertial, I get a segmentation fault in approximately the same place at every execution. 

I am using a Linux Mint 19 (`Tara') Cinnamon partition of an HP Pavilion 15-p209na with an Intel© Core™ i3-5010 (2 cores @ 2.10GHz) and 8 GB RAM. I'm running ROS melodic (ROS version 1.4.10).

I run `roscore` in one tab, `rosrun ORB_SLAM3 Mono_Inertial ./Vocabulary/ORBvoc.txt ./Examples/Monocular-Inertial/midair.yaml` in a second tab, and `rosbag play --pause <PATH TO .BAG>` in the third tab. 

I've pasted the .yaml setup file that I'm using at the bottom of this post. I will note that I'm unclear what Tbc is - on the .yaml comment it is `camera to body`, but in the publication it is `body to camera`. I've tried using both versions of the transformation but both encounter the same issue. I'll also mention that as this is a synthetic dataset, I've been able to tune the IMU parameters really finely and - so long as I've implemented the noise implementation correctly - the parameters should be a good representation of the data.

Execution starts fine, and things seem to initialise okay with a couple of tries (snippet with repetition removed is shown below), and then it eventually manages to get a track on the map. But then not long after the IMU initialisation, I get a 'RESETING FRAME!!!' message followed immediately by the seg fault (also shown below).

<h1>STDOUT</h1>

```
First KF:0; Map init KF:0
New Map created with 159 points
Point distribution in KeyFrame: left-> 159 --- right-> 0
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 1
mnInitialFrameId = 0
2 Frames set to lost
...
First KF:6; Map init KF:3
New Map created with 375 points
Point distribution in KeyFrame: left-> 375 --- right-> 0
build optimization graph
start optimization
end optimization
IMU in Map 0 is initialized
RESETING FRAME!!!
Segmentation fault (core dumped)
```

I've run gdb with this to try and trace things back by running `rosrun --prefix 'xterm -e gdb --args' ORB_SLAM3 Mono_Inertial ./Vocabulary/ORBvoc.txt ./Examples/Monocular-Inertial/midair.yaml` in the second terminal. I've included the SIGSEGV basic error below, and it <b>looks like</b> (to me, someone who isn't very familiar with Eigen or g2o) that it's the third-party stuff that is struggling, though I assume it is in some way related to my setup (since people are reporting that other datasets work just fine, and no-one that I've seen is having this particular issue).

<h1>GDB SHORT</h1>

```
Thread 9 "Mono_Inertial" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb37fe700 (LWP 8078)]
0x00007ffff6a157a0 in void Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, 3, 3, 0, 3, 3>&, Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::internal::assign_op<double, double> const&) ()
   from /home/rory_haggart/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#0  0x00007ffff6a157a0 in void Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, 3, 3, 0, 3, 3>&, Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::internal::assign_op<double, double> const&) ()
    at /home/rory_haggart/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
```

The full gdb backtrace is below, which doesn't help me at all but maybe someone will spot something...

<h1>GDB LONG</h1>

```
#1  0x00007ffff6ac39b6 in ORB_SLAM3::EdgePriorPoseImu::EdgePriorPoseImu(ORB_SLAM3::ConstraintPoseImu*) ()
    at /home/rory_haggart/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#2  0x00007ffff69fc6cf in ORB_SLAM3::Optimizer::PoseInertialOptimizationLastFrame(ORB_SLAM3::Frame*, bool) ()
    at /home/rory_haggart/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#3  0x00007ffff6944a99 in ORB_SLAM3::Tracking::TrackLocalMap() ()
    at /home/rory_haggart/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#4  0x00007ffff6947cc8 in ORB_SLAM3::Tracking::Track() ()
    at /home/rory_haggart/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#5  0x00007ffff694a651 in ORB_SLAM3::Tracking::GrabImageMonocular(cv::Mat const&, double const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
    at /home/rory_haggart/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#6  0x00007ffff6921069 in ORB_SLAM3::System::TrackMonocular(cv::Mat const&, double const&, std::vector<ORB_SLAM3::IMU::Point, std::allocator<ORB_SLAM3::IMU::Point> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
    at /home/rory_haggart/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#7  0x00005555555665b1 in ImageGrabber::SyncWithImu() ()
#8  0x00007ffff5793733 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff78296db in start_thread (arg=0x7fffb37fe700)
    at pthread_create.c:463
#10 0x00007ffff51eda3f in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

<h1>YAML FILE</h1>

```
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 512.0
Camera.fy: 512.0 
Camera.cx: 512.0
Camera.cy: 512.0

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

# Camera resolution
Camera.width: 1024
Camera.height: 1024

# Camera frames per second 
Camera.fps: 25

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Transformation from camera to body-frame (imu)
Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [0.0, 0.0, 1.0, 0.0,
          1.0, 0.0, 0.0, 0.0,
          0.0, 1.0, 0.0, 0.0,
          0.0, 0.0, 0.0, 1.0]

# IMU noise
IMU.NoiseGyro: 0.004000000  #1.6968e-04 
IMU.NoiseAcc: 0.080000000  #2.0e-3
IMU.GyroWalk: 0.000002000
IMU.AccWalk: 0.000040000 # 3e-03
IMU.Frequency: 100

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 2000 # 1000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -3.5 # -1.8
Viewer.ViewpointF: 500
```

siddharthcb, 2021-06-17:
maybe reduce the camera resolution and try

Wanghy118, 2022-04-20:
hi! I am also trying to run the [MidAir]dataset on ORB-SLAM3 Mono_Inertial.  I met the "Segmentation fault (core dumped)"  after "new map created with points!":
Loading images for sequence 0...LOADED!
Loading IMU for sequence 0...LOADED!

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial
Loading settings from ./Monocular-Inertial/Midair.yaml
Camera1.k3 optional parameter does not exist...
	-Loaded camera 1
	-Loaded image info
IMU.InsertKFsWhenLost optional parameter does not exist...
	-Loaded IMU calibration
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 512 512 512 512 ]
	-Camera 1 distortion parameters: [  0 0 0 0 ]
	-Original image size: [ 1024 , 1024 ]
	-Current image size: [ 600 , 350 ]
	-Camera 1 parameters after resize: [  300 175 300 175 ]
	-Sequence FPS: 25
	-Gyro noise: 0.00016999999934341758
	-Accelerometer noise: 0.0020000000949949026
	-Gyro walk: 1.9392999092815444e-05
	-Accelerometer walk: 0.0030000000260770321
	-IMU frequency: 100
	-Features per image: 1000
	-ORB scale factor: 1.2000000476837158
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.Starting the Viewer
First KF:0; Map init KF:0
New Map created with 132 points
Segmentation fault (core dumped)

Wanghy118, 2022-04-20:
Hi， I am trying to run the [MidAir](https://midair.ulg.ac.be/index.html) dataset on ORB-SLAM3 Mono_Inertial, too! I can run Mono_Inertial  successfully, but IMU can not be initialized! Have you ever met this problem? could you help me? Thanks!

khoj-pez, 2023-07-27:
Did you end up figuring this out? I'm running into a similar issue.