UZ-SLAMLab/ORB_SLAM3 #313: Mono-Inertial with aerial images: Fail to track local map!

NaiaraES, 2021-05-10:
I am trying to run the Mono-Inertial algorithm with an aerial dataset registered by a fixed-wing airplane with the camera pointing downwards. The problem is that the track is lost at almost every step:

First KF:0; Map init KF:0
New Map created with 529 points
Point distribution in KeyFrame: left-> 529 --- right-> 0
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
IMU in Map 0 is initialized
Fail to track local map!
IMU is not or recently initialized. Reseting active map..
....

At the beginning I thought that the problem was the Tbc matrix, but I have checked it several times:
  - IMU/BODY: Xb front of the plane, Yb right wing, Zb downwards
  - CAMERA: Xc right wing, Yc tail of the plane, Zc downwards

So the matrix rotation from camera to IMU correspondsbto  a rotation of -90º over Z axis, that is:
Rbc =  0.0, -1.0,  0.0; 1.0,  0.0,  0.0; 0.0,  0.0,  1.0]
I also tested using the inverse (just in case), but the same result.

Looking at the examples (EuroC and TUM) I released that all of them use images pointing forward. I couldn't find any example using aerial images, like in my case. So, it is required any additional rotation to the images or IMU measurements since it is not pointing forward? If not, do anyone have any idea what is happening? I am able to run the examples even the Mono algorithm with my dataset (it shows a great error in Z axis). The problem is when I try to introduce the inertial data.

ghkeller, 2021-06-17:
I had the same "fail to track local map" issue with my indoor aerial dataset. I used Kalibr (https://github.com/ethz-asl/kalibr) to get the transform from my camera to the IMU which matches my derived values. I also was able to make it work with only mono images, but when I include IMU it fails. It seems as though there might be some problem with scaling because when the point map begins to form, after about a second it rapidly shrinks down and then fails.

BartKevers, 2021-08-05:
> I have the same "fail to track local map" issue with my indoor aerial dataset. I used Kalibr (https://github.com/ethz-asl/kalibr) to get the transform from my camera to the IMU which matches my derived values. I also was able to make it work with only mono images, but when I include IMU it fails. It seems as though there might be some problem with scaling because when the point map begins to form, after about a second it rapidly shrinks down and then fails.

I have a similar problem which I think is related to the initial scaling before IMU initialisation. did you manage to find a solution?

Huyhust13, 2022-09-29:
Hi @BartKevers @ghkeller , 
I have a similar problem. Did you resolve this?