UZ-SLAMLab/ORB_SLAM3 #315: Fails to build V0.4: LocalMapping.cc & KannalaBrandt8.cc (no match for 'operator/')

y-elena, 2021-05-10:
Hi, thanks for the great work! I was previously using ORB_SLAM3 V0.3 with the Monocular-Inertial node in ROS. and now want to compile the new version...

I am working with ROS Melodic, Ubuntu 18.04, Python 2.7 and have all the dependencies installed... one known difference is the eigen3 version: 3.3.4. 

Here is the output:
`CMakeFiles/ORB_SLAM3.dir/build.make:110: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o' failed`

caused by this error:

`.../ORB_SLAM3/src/LocalMapping.cc:629:49: error: no match for ‘operator/’ (operand types are ‘cv::Matx<float, 3, 1>’ and ‘float’)
                 x3D = x3D_h.get_minor<3,1>(0,0) / x3D_h(3);`

and

`...ORB_SLAM3/src/CameraModels/KannalaBrandt8.cpp:534:41: error: no match for ‘operator/’ (operand types are ‘cv::Matx<float, 3, 1>’ and ‘float’)
         x3D = x3D_h.get_minor<3,1>(0,0) / x3D_h(3);`

Does anyone have any suggestions? to fix the compilation error?

SIDE NOTE: I followed [this](https://icode9.com/content-3-961807.html) which suggested commenting out line 534 in KannalaBrandt8.cpp and replacing `void LocalMapping::CreateNewMapPoints()` function with the same one from the previous version (V0.3). It now compiles but...

When I run ORB-SLAM3 in Mono_Inertial mode (`rosrun ORB_SLAM3 Mono_Inertial .../ORB_SLAM3/Vocabulary/ORBvoc.txt ...*.yaml`) it immediately crashes:

[image]

So don't try this ^^

hellovuong, 2021-05-10:
you may change  
`x3D = x3D_h.get_minor<3,1>(0,0) / x3D_h(3);`
to 
`x3D = x3D_h.get_minor<3,1>(0,0) (1/x3D_h(3));`
as suggest in #312

aravindSolteq, 2021-05-11:
#300 Refer this

y-elena, 2021-05-11:
> #300 Refer this

@aravindSolteq Thank you, this solution worked for me.

closed 2021-12-23 by jdtardos