UZ-SLAMLab/ORB_SLAM3 #316: run RGBD with Realsense D435 to optimizer.optimize(its[it]); (Optimizer.cc). Segmentation fault ;

fsc1023, 2021-05-11:
` for (size_t it = 0; it < 4; it++) {`
`    std::cout << "--pFrame->mTcw : " << pFrame->mTcw << std::endl;`
`    vSE3->setEstimate(Converter::toSE3Quat(pFrame->mTcw));`
`    std::cout << "--tag 1" << std::endl;`
`    optimizer.initializeOptimization(0);`
`    std::cout << "--its[" << it << "] : " << its[it] << std::endl;`
`    optimizer.optimize(its[it]);  // come to this line, Segmentation fault`
`    std::cout << "--tag 3" << std::endl;`
`...`
`}`
but running  Stereo, have no error!

**logs display:**
`
`--pFrame->mTcw : [1, 0, 0, 0;` 
`  0, 1, 0, 0; 
`  0, 0, 1, 0; `
`  0, 0, 0, 1] `
` --tag 1 `
` --its[0] : 10 `
` Segmentation fault (core dumped)`

Please help me.
Thank you very much.

hellovuong, 2021-05-13:
Try to remove CMakeLists.txt in both ORB_SLAM3 and g2o "march=native" Optimization options，It may be helpful for you.

fsc1023, 2021-05-17:
> Try to remove "march=native" Optimization options in CMakeLists.txt of both ORB_SLAM3 and g2o ，It may be helpful for you.

Thank you very much, it running successfully. Thank you for your suggestion!