UZ-SLAMLab/ORB_SLAM3 #317: Failed to load image

skewed666, 2021-05-12:
why i get this error：
book@shadow:~/Dev/ORB_SLAM3$ ./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml ~/Datasets/EuRoc/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono
num_seq = 1
file name: dataset-MH01_mono
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 458.65399169921875
- fy: 457.29598999023438
- cx: 367.21499633789062
- cy: 248.375
- k1: -0.28340810537338257
- k2: 0.073959067463874817
- p1: 0.00019359000725671649
- p2: 1.7618711353861727e-05
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Failed to load image at: /home/book/Datasets/EuRoc/MH01/mav0/cam0/data/1403636579763.png84

yuguanfeng, 2021-06-09:
I have met the same problem. Do you solve it ?

yuguanfeng, 2021-06-10:
I find the same problem in ORB2 Issue #903. It seems like there was an invisible character '\r ' in the timestamp file. I create a new blank MH01.txt. Then copy all from /EuRoC_TimeStamps/MH01.txt to my MH01.txt and replace the original one. It solved!

marcelosemo, 2021-11-08:
@skewed666 Did you solve your problem? I had the exact same issue, and it originated from a very stupid problem: The Machine Hall Dataset one can download from [](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets) is called "MH_01_easy", so when I renamed it, I renamed it to "MH_01" instead of "MH01", which is why the images couldn't be found.

yriuss, 2021-11-11:
@marcelosemo I tried rename to "MH01" but it didn't solve. The output is:

```
Launching MH01 with Monocular sensor
num_seq = 1
file name: dataset-MH01_mono
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 458.65399169921875
- fy: 457.29598999023438
- cx: 367.21499633789062
- cy: 248.375
- k1: -0.28340810537338257
- k2: 0.073959067463874817
- p1: 0.00019359000725671649
- p2: 1.7618711353861727e-05
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Failed to load image at: ~/data/EuRoC/MH01/mav0/cam0/data/1403636579763555584.png
```

The path is: "~/data/EuRoC/MH01/mav0/cam0/data".

I tried the @yuguanfeng solution ass well, but I didn't see the invisible '\r' character... Anybody can help me, please?

yriuss, 2021-11-11:
@yuguanfeng solution solved for me. Thanks! I used notepad++ to extract only the useful characters.

xiaomofang, 2022-06-10:
I also met the same problem, you can try to check the path again,for example,~/Datasets/EuRoc/MH01,should be ./Datasets/EuRoc/MH01,Double-check again