UZ-SLAMLab/ORB_SLAM3 #324: what():  Pangolin X11: Failed to create an OpenGL context Aborted

slypanzer, 2021-05-24:
when I release ORB_SLAM3, an error occured. And the complete information as follows:
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Failed to create an OpenGL context
Aborted

Could somone help me please?

KevinYuan666, 2021-06-10:
Has it been solved？

AlanWu857, 2021-12-03:
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Failed to open X display
./tum_rgbd.sh：行 53: 1650060 已放弃               （核心已转储）

HJMGARMIN, 2022-09-20:
You can update the NVIDIA driver and reinstall Pangolin to solve it.

EmmanuelMess, 2024-02-19:
See #486.