UZ-SLAMLab/ORB_SLAM3 #325: Scale initialization in  InitializeIMU()

vitalemonate, 2021-05-24:
I found that the initialization method of the scale in the code and the paper is different. In [_Inertial-Only Optimization for Visual-Inertial Initialization_](https://arxiv.org/pdf/2003.05766.pdf), the scale is initialized as follows: 

[image]

but scale is directly initialized to 1.0 in InitializeIMU(), which is in LocalMapping.cc

[image]

I have two questions:
1. How to understand the initialization method of the scale that correspond to median scene depth of 1, 4 and 16 meters?
2. Is the scale directly initialized to 1.0 for the consideration of the amount of calculation?

Thanks for your answers!

BartKevers, 2021-08-05:
I have similar questions. Did you manage to find answers?

vitalemonate, 2021-12-03:
> 

Not yet

vitalemonate, 2021-12-03:
Not yet

> I have similar questions. Did you manage to find answers?