UZ-SLAMLab/ORB_SLAM3 #326: Segmentation Fault:  __strlen_avx2 () at ../sysdep/x86_64/multiarch/strlen-avx2.S:96

guglielmodev, 2021-05-24:
When I run ROS Stereo or Stereo_Inertial nodes they crash soon after start

```
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo
Segmentation fault (core dumped)
```

Seems similar to #286 but with gdb I get:

```
 __strlen_avx2 () at ../sysdep/x86_64/multiarch/strlen-avx2.S:96
96: ../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.
```

guglielmodev, 2021-05-24:
Can it be related to the OpenCV warnings at the end of ros_build?

```
/usr/bin/x86_64-linux-gnu-ld: warning: libopencv_highgui.so.4.5, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_highgui.so.3.2
/usr/bin/x86_64-linux-gnu-ld: warning: libopencv_calib3d.so.4.5, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_calib3d.so.3.2
/usr/bin/x86_64-linux-gnu-ld: warning: libopencv_features2d.so.4.5, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_features2d.so.3.2
/usr/bin/x86_64-linux-gnu-ld: warning: libopencv_imgproc.so.4.5, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_imgproc.so.3.2
/usr/bin/x86_64-linux-gnu-ld: warning: libopencv_core.so.4.5, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_core.so.3.2
```

Jinyu-M, 2021-06-21:
I got the same problem....Do your find any solutions? @guglielmodev

Alkrick, 2021-11-06:
I have the same problem, has anyone found a solution? @guglielmodev @Jinyu-M

mateosss, 2021-11-06:
I remember having this problem when not compiling all dependencies with the same flags. In particular, not compiling them with the same `-march=native` flag.

Chenliezhen259, 2021-11-07:
> I remember having this problem when not compiling all dependencies with the same flags. In particular, not compiling them with the same `-march=native` flag.

it do not work.