UZ-SLAMLab/ORB_SLAM3 #327: Fail to compile

KKKKKyle, 2021-05-25:
I'm following steps to compile ORB_SLAM3 using "./build.sh" after installing all the prerequisites (eigen3.2.1) and I encountered this issue on Ubuntu18.04:

`-- Configuring done
-- Generating done
-- Build files have been written to: /home/kyle/ORB_SLAM3/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- BUILD TYPE:Release
-- Compiling on Unix
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenMP_C: -fopenmp  
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenMP_CXX: -fopenmp  
-- Found OpenMP: TRUE   
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kyle/ORB_SLAM3/Thirdparty/g2o/build
Scanning dependencies of target g2o
[  3%] Building CXX object CMakeFiles/g2o.dir/g2o/types/types_sba.cpp.o
In file included from /home/kyle/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_sba.h:30:0,
                 from /home/kyle/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_sba.cpp:27:
/home/kyle/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_vertex.h:62:74: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
 ypedef Eigen::Map<Matrix<double, D, D>, Matrix<double,D,D>::Flags & AlignedBit ? Aligned : Unaligned >  HessianBlockType;
                                                                     ^~~~~~~~~~
In file included from /usr/include/eigen3/Eigen/Core:344:0,
                 from /home/kyle/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30,
                 from /home/kyle/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41,
                 from /home/kyle/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/kyle/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_sba.h:30,
                 from /home/kyle/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_sba.cpp:27:
[... 588 lines cut ...]
[ 60%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/MapDrawer.cc.o
[ 62%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Initializer.cc.o
[ 64%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/KeyFrameDatabase.cc.o
[ 59%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/MLPnPsolver.cpp.o
[ 60%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/OptimizableTypes.cpp.o
[ 63%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/G2oTypes.cc.o
[ 65%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/CameraModels/KannalaBrandt8.cpp.o
[ 67%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/TwoViewReconstruction.cc.o
[ 68%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/CameraModels/Pinhole.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Thirdparty/g2o/CMakeFiles/g2o.dir/build.make:518: recipe for target 'Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_with_hessian.cpp.o' failed
make[2]: *** [Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_with_hessian.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Thirdparty/g2o/CMakeFiles/g2o.dir/build.make:182: recipe for target 'Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/marginal_covariance_cholesky.cpp.o' failed
make[2]: *** [Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/marginal_covariance_cholesky.cpp.o] Error 4
`

Many thanks in advance for any suggesiton!

gotiobg, 2021-05-31:
> c++: internal compiler error: Killed (program cc1plus)

This means that there isn't enough RAM available to finish making.
Check here for resolution https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/118#issuecomment-726648539

ravilakhotia2006, 2022-01-19:
@Kyle0936 - were you able to resolve this issue ? I am getting the same errors and removing -j from make -j in build.sh doesn't fix anything.