UZ-SLAMLab/ORB_SLAM3 #328: macOS build issue

wcandres, 2021-05-25:
Hello! 

Relevant info:
- macOS Big Sur 11.3.1 (20E241)
- Made a couple of file changes for building
  - build.sh
    - Added `-DCMAKE_CXX_COMPILER=/usr/local/bin/g++-10 -DCMAKE_C_COMPILER=/usr/local/bin/gcc-10` to each cmake call to use gcc/g++ instead of default clang
  - Each CMakeLists.txt file
    - Was failing to detect boost library
    - Added `FIND_PACKAGE(Boost)` and appended `${Boost_INCLUDE_DIR}` to `include_directories`

The most relevant part of the error is as follows, but I can decipher no meaning from the error.
```
make[2]: *** No rule to make target `../Thirdparty/DBoW2/lib/libDBoW2.so', needed by `../lib/libORB_SLAM3.dylib'.  Stop.
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [all] Error 2
```

Any help with this issue would be greatly appreciated.

Here is a full log from a fresh build: 
[freshbuild.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/6542750/freshbuild.txt)

And a shorter log from a build attempt after a fresh build:
[build.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/6542749/build.txt)

gotiobg, 2021-05-31:
In this freshbuild.txt log file, there is this at the beginning:
>   MACOSX_RPATH is not specified for the following targets:
>   DBoW2

I suppose you have to add the path to DBoW2 to your MACOSX_RPATH environment variable. I'm not familiar with macOS but in Windows, you have to append it here (far right):
[image]

Or maybe you have to provide a path in build.sh which links to CMakeLists.txt. If this is the case, then you might need to add "add_subdirectory(Thirdparty/DBoW2)" or edit line "${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.so".

BigJohnn, 2021-06-02:
> Hello!
>
> Relevant info:
>
> * macOS Big Sur 11.3.1 (20E241)
> * Made a couple of file changes for building
>
>   * build.sh
>
>     * Added `-DCMAKE_CXX_COMPILER=/usr/local/bin/g++-10 [...]

1.modify CMakeLists in DBoW2/g2o, make sure there dylibs are generated, e.g. ~/Codes/ORB_SLAM3/Thirdparty/DBoW2/lib/libDBoW2.dylib
2.modify /ORBSLAM3_someroot/CMakeLists.txt
${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.so
${PROJECT_SOURCE_DIR}/Thirdparty/g2o/lib/libg2o.so 
to
${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.dylib
${PROJECT_SOURCE_DIR}/Thirdparty/g2o/lib/libg2o.dylib
Good Luck!

dandingol03, 2022-11-23:
> ORBSLAM3_someroot

no use

gabriel-bronfman, 2023-10-25:
Not sure if anyone is still checking this. My dylib file for DBow2 is empty. I cannot seem to make a dylib. Has anyone made any headway here?