UZ-SLAMLab/ORB_SLAM3 #330: Are jacobians wrong in G2oTypes.cc?

leavesnight, 2021-06-01:
we found computeError uses bprior - VA/VG->estimate() but linearizeOplus() gives Identity() not -I, is this part wrong?
`void computeError(){        const VertexAccBias* VA = static_cast&lt;const VertexAccBias*&gt;(_vertices[0]);        _error = bprior - VA-&gt;estimate();    }

void EdgePriorAcc::linearizeOplus(){    // Jacobian wrt bias    _jacobianOplusXi.block&lt;3,3&gt;(0,0) = Eigen::Matrix3d::Identity();
}
void EdgePriorGyro::linearizeOplus(){    // Jacobian wrt bias    _jacobianOplusXi.block&lt;3,3&gt;(0,0) = Eigen::Matrix3d::Identity();
}`

treememory, 2021-06-12:
maybe a wrong. it goes better after i add -:
_jacobianOplusXi.block<3,3>(0,0) = -Eigen::Matrix3d::Identity();

sherman-luo, 2022-12-05:
Agreed, this is incorrect. I unit tested this code and it the bias will not converge properly