UZ-SLAMLab/ORB_SLAM3 #333: run Mono_Inertial Segmentation fault (core dumped)

liangchen97, 2021-06-04:
nvidia@nvidia-desktop:~/catkin_ws/src/ORB_SLAM3$ rosrun ORB_SLAM3 Mono_Inertial /Vocabulary/ORBvoc.txt /Examples/Monocular-Inertial/EuRoC.yaml

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Segmentation fault (core dumped)

yuguanfeng, 2021-06-09:
I have met the same problem when I try to rosrun Mono node. Did you solve it?

yuguanfeng, 2021-06-10:
It may be resulted from the different versions of OpenCV used in ./build.sh and ./build_ros.sh. In ORB_SLAM3/Thirdparty/DBoW2/CMakeLists.txt and ORB_SLAM3/CMakeLists.txt, it requires opencv 4. But in ORB_SLAM3/Examples/ROS/ORB_SLAM3/CMakeLists.txt, it requires opencv 3. I have installed opencv4.1.0 outside and opencv3.3.1 in ros. Finally it leads to the segmentation fault. To solve it, modify CMakeLists.txt to require opencv 3.

liangchen97, 2021-07-02:
谢谢老哥，可以使用了，不过我用realsense d455跑Mono_Inertial节点效果很差，不知道你试过没

Cheny5863, 2021-08-31:
> It may be resulted from the different versions of OpenCV used in ./build.sh and ./build_ros.sh. In ORB_SLAM3/Thirdparty/DBoW2/CMakeLists.txt and ORB_SLAM3/CMakeLists.txt, it requires opencv 4. But [...]

老哥稳