UZ-SLAMLab/ORB_SLAM3 #334: building issue

jinchen62, 2021-06-07:
got this error when building orb_slam3 on ubuntu 20.04, I have installed glew and built pangolin successfully.

Any idea or suggestion on fixing it?

> [ 45%] Linking CXX shared library ../lib/libORB_SLAM3.so
/usr/bin/ld: cannot find -llibglew_static
/usr/bin/ld: cannot find -llibglew_shared
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:559: ../lib/libORB_SLAM3.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:268: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

johnc2021, 2021-06-13:
Have you've tried `bash build.sh` and scrolling up to see what errors you are actually getting?

jinchen62, 2021-06-14:
@johnc2021 I think there was no other error. I built it on a 18.04 system with the same setup, so maybe 20.04 is not compatible. Thanks for the reply!