UZ-SLAMLab/ORB_SLAM3 #335: WAITING FOR IMAGES error tested with IntelRealsense T265 and D435i

pi1ing, 2021-06-08:
Hi,
 I have successfully run ORB-SLAM3 with EuRoC Dataset and ROS example. I want to run ROS ORB-SLAM3 with IntelRealsense T265 & D435i,  by code like this in three different terminals(tested with D435i):

> roscore
> roslaunch realsense2_camera rs_camera.launch
> rosrun ORB_SLAM3 Mono /home/shen/work/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/shen/work/ORB_SLAM3/yaml/D435i.yaml 

here is my D435i.yaml:

> %YAML:1.0
> 
> #--------------------------------------------------------------------------------------------
> # Camera Parameters. Adjust them!
> #--------------------------------------------------------------------------------------------
> 
> # Camera calibration and distortion parameters (OpenCV) 
> Camera.fx: 615.9417724609375
> Camera.fy: 616.0935668945312
> Camera.cx: 322.3533630371094
> Camera.cy: 240.44674682617188
> 
> Camera.k1: 0.0
> Camera.k2: 0.0
> Camera.p1: 0.0
> Camera.p2: 0.0
> Camera.p3: 0.0
> 
> Camera.width: 640
> Camera.height: 480
> 
> # Camera frames per second 
> Camera.fps: 30.0
> 
> # IR projector baseline times fx (aprox.)
> # bf = baseline (in meters) * fx, D435i的 baseline = 50 mm 
> Camera.bf: 30.797   
> 
> # Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
> Camera.RGB: 1
> 
> # Close/Far threshold. Baseline times.
> ThDepth: 40.0
> 
> # Deptmap values factor
> DepthMapFactor: 1000.0
> 
> #--------------------------------------------------------------------------------------------
> # ORB Parameters
> #--------------------------------------------------------------------------------------------
> 
> # ORB Extractor: Number of features per image
> ORBextractor.nFeatures: 1000
> 
> # ORB Extractor: Scale factor between levels in the scale pyramid 	
> ORBextractor.scaleFactor: 1.2
> 
> # ORB Extractor: Number of levels in the scale pyramid	
> ORBextractor.nLevels: 8
> 
> # ORB Extractor: Fast threshold
> # Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
> # Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
> # You can lower these values if your images have low contrast			
> ORBextractor.iniThFAST: 20
> ORBextractor.minThFAST: 7
> 
> #--------------------------------------------------------------------------------------------
> # Viewer Parameters
> #--------------------------------------------------------------------------------------------
> Viewer.KeyFrameSize: 0.05
> Viewer.KeyFrameLineWidth: 1
> Viewer.GraphLineWidth: 0.9
> Viewer.PointSize: 2
> Viewer.CameraSize: 0.08
> Viewer.CameraLineWidth: 3
> Viewer.ViewpointX: 0
> Viewer.ViewpointY: -0.7
> Viewer.ViewpointZ: -1.8
> Viewer.ViewpointF: 500
> 

and the terminal result is like this:

> ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
> ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
> This program comes with ABSOLUTELY NO WARRANTY;
> This is free software, and you are welcome to redistribute it
> under certain conditions. See LICENSE.txt.
> 
> Input sensor was set to: Monocular
> 
> Loading ORB Vocabulary. This could take a while...
> Vocabulary loaded!
> 
> Creation of new map with id: 0
> Creation of new map with last KF id: 0
> Seq. Name: 
> 
> Camera Parameters: 
> *Not Supported Camera Sensor*
> Check an example configuration file with the desired sensor
> - fps: 30
> - color order: RGB (ignored if grayscale)
> 
> ORB Extractor Parameters: 
> - Number of Features: 1000
> - Scale Levels: 8
> - Scale Factor: 1.2
> - Initial Fast Threshold: 20
> - Minimum Fast Threshold: 7
> 

while the image windows shows in a black screen with message "WAITING FOR IMAGES".
I have tested IntelRealsense_ros like:

> rosrun image_view image_view  image:=/camera/color/image_raw
shows image, which means the intelrealsense_ros build successfully.

Looking forward to your help!

ardabbour, 2021-06-09:
You have to remap the topics that the ROS nodes are listening to as such:
`rosrun ORB_SLAM3 [node name] [path to vocabulary file] [path to settings file] /camera/image_raw:=/camera/color/image_raw  /imu:=/camera/imu`

pi1ing, 2021-06-11:
@ardabbour 
Thank you for your reply! Although I still have difficult in running Mono_Inertial, and Stereo_Inertial mode for I don't know quite much about ROS, the Mono mode could run successfully by your way. Anyway, thank you for your help!

closed 2021-06-11 by pi1ing