UZ-SLAMLab/ORB_SLAM3 #336: Wrong direction of gravity in the world coordinate system

treememory, 2021-06-12:
thank you for your work~

treememory, 2021-06-12:
the result trajectory in world coordinate is worng?

treememory, 2021-06-12:
There seems to be a problem with the gravity direction of the output trajectory. The trajectory of Euroc's MH05 data is compared with VINS. The result is as follows. The gravity direction of VINS (color trajectory) is correct, but the trajectory line of ORBSLAM3 (white trajectory) is obviously wrong. Why, isn’t there a gravity direction estimation and optimization?
[image]

treememory, 2021-06-12:
There seems to be a problem with the gravity direction of the output trajectory. The trajectory of Euroc's V203 data is compared with VINS. The result is as follows. The gravity direction of VINS (color trajectory) is correct, but the trajectory line of ORBSLAM3 (white trajectory) is obviously wrong. Why, isn’t there a gravity direction estimation and optimization?
[image]

xinruozhishui201314, 2021-06-16:
你好，请问这个问题查的怎么样？