UZ-SLAMLab/ORB_SLAM3 #339: A docker image with everything installed? Or better documented install instructions?

NickPerezCarletonUniversity, 2021-06-19:
I spent a few hours trying to get orb-slam 3 to run on Ubuntu 20.04 yesterday. I had no success. Is there a better way to package your program? Such as in a docker image that contains all the required dependencies?... That way you just have to download the docker image and you are ready to go. Your program has a huge amount of dependencies, making it quite brittle.

Or at the very least, the installation instructions could be more straightforward and better documented. For example, someone could write out in order all the required Linux commands to install all the dependencies with the required versions.

partus, 2021-06-28:
A dockerfile would be very appreciated.  I tried to build one with replacing  FROM borda/docker_python-opencv-ffmpeg:py3.6 in  [yuyou/ORB_SLAM2](https://github.com/yuyou/ORB_SLAM2/blob/master/docker/Dockerfile.ubuntu18) dockerfile , which seems to be broken due to unavailable image with [this](https://github.com/PacktPublishing/OpenCV-4-with-Python-Blueprints-Second-Edition/blob/master/dockerfiles/gpu.Dockerfile) and it worked fine with ORB_SLAM2. 

The built of  ORB_SLAM3 fails. 

Seem like the main errors are like:
`/ORB_SLAM3/src/LocalMapping.cc:628:49: error: no match for ‘operator/’ (operand types are ‘cv::Matx<float, 3, 1>’ and ‘float’)
                 x3D = x3D_h.get_minor<3,1>(0,0) / x3D_h(3);`

No matter if it's opencv4 which I used first, or 3 .

I think it would help a lot anyone who wants to experiment with the project or run it on some new machine if you could provide a clear dockerfile with right dependencies and right versions so that everything builds seamlessly with it.

marcusabate, 2021-08-30:
Also voicing my support for a dockerfile. I recently have been trying to get the system working natively on a 20.04 system, which has failed mostly at the Pangolin level. I think either some recent updates to Pangolin broke the system, or other system-wide packages aren't playing well with it. I tried installing natively on 18.04 and saw the same issues. Related to #374.

I've been working on a docker file to run the ROS nodes, which is able to build Pangolin and get to the ORB build script, but it fails there. Fails whether running `build.sh` or `build_ros.sh`. Here is what I have so far (missing clean-up):

```
FROM ros:melodic-ros-base

RUN apt-get update && apt-get install -y --no-install-recommends apt-utils
RUN apt-get install -y \
        # Base tools
        cmake \
        build-essential \
        git \
        unzip \
        pkg-config \
        # OpenCV dependencies
        python-numpy \
        # Pangolin dependencies
        libgl1-mesa-dev \
        libglew-dev \
        libpython2.7-dev \
        libeigen3-dev
        # ORB-SLAM3 dependencies
        # ROS packages

RUN mkdir -p /dependencies/src/

# Build OpenCV
RUN cd /dependencies/src && git clone https://github.com/opencv/opencv.git && \
    cd opencv && \
    git checkout 4.2.0 && \
    mkdir build && cd build && \
    cmake -D CMAKE_BUILD_TYPE=Release -D CMAKE_INSTALL_PREFIX=/usr/local .. && \
    make -j3 && make install

# Build Pangolin
RUN cd /dependencies/src && git clone https://github.com/stevenlovegrove/Pangolin && \
    cd Pangolin && mkdir build && cd build && cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-std=c++11 .. && \
    make -j$nproc && make install

# Build ORB-SLAM3 for ROS
RUN mkdir -p /catkin_ws/src/
RUN git clone https://github.com/UZ-SLAMLab/ORB_SLAM3 /catkin_ws/src/ORB_SLAM3
RUN . /opt/ros/melodic/setup.sh && \
    export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/catkin_ws/src/ORB_SLAM3/Examples/ROS && \
    cd /catkin_ws/src/ORB_SLAM3/ && \
    chmod +x build_ros.sh && chmod +x build.sh && \
    ./build.sh && ./build_ros.sh
```

marcusabate, 2021-08-30:
@NickPerezCarletonUniversity @partus over on the Pangolin side the maintainer recommended that for ORB we use tag v0.6. This worked well for me. I've edited the dockerfile above, it builds on my machine.

Note that the umerged PR you mentioned compiles but probably has a logic error. It's described in a comment on the PR.

jahaniam, 2021-09-01:
Hi guys, I have written a docker for ros melodic if you are interested.

https://github.com/jahaniam/orbslam3_docker

NickPerezCarletonUniversity, 2021-09-01:
@jahaniam i noticed you are doing some nvidia commands in your docker container. is that to allow GPU access? also, did you make this container for Windows 10?... (sorry for noob questions, I am a bit of a docker noob at the moment)

jahaniam, 2021-09-01:
@NickPerezCarletonUniversity Good point. I used Nvidia docker container for the GUI to work properly and my laptop has gpu. If you don't have GPU you can remove those commands. It was built in Linux. I'm not sure about windows 10 but I assume it should work.

stellarpower, 2021-09-04:
I'm also working on a container image (without the ROS extras on top) and would happily contribute back.

NickPerezCarletonUniversity, 2021-09-04:
@stellarpower that's awesome, keep us posted! :) will your docker being running on a windows system? I haven't played around with GPU access in Docker running on a Windows system, but it is on my backlog of things to do.

UcefMountacer, 2021-09-13:
> @NickPerezCarletonUniversity @partus over on the Pangolin side the maintainer recommended that for ORB we use tag v0.6. This worked well for me. I've edited the dockerfile above, it builds on my machine.

Hi,

I did the same, install pangolin v0.6 and other packages with respect to the versions required. I did that on colab and ubuntu 20 but it does not work. The build fails for some reason:

```
CMakeFiles/ORB_SLAM3.dir/build.make:88: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o] Error 1
CMakeFiles/Makefile2:109: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

I get a lot of errors like this related to LocalMapping.cc file :

```
/content/ORB_SLAM3/src/LocalMapping.cc: In member function ‘void ORB_SLAM3::LocalMapping::ScaleRefinement()’:
/content/ORB_SLAM3/src/LocalMapping.cc:1455:15: warning: unused variable ‘N’ [-Wunused-variable]
     const int N = vpKF.size();
               ^
```

jahaniam, 2021-09-14:
@NickPerezCarletonUniversity @stellarpower @UcefMountacer  I changed and fixed some problems with the nvidia drivers. Feel free to retest. It enables the UI properly https://github.com/jahaniam/orbslam3_docker. 

Also the fix for local mapping for operato / not found, I have done it in https://github.com/jahaniam/ORB_SLAM3/tree/docker_opencv3.2_fix  

I notice this fix is not required if you use opencv v4 but that's untested.

UcefMountacer, 2021-09-14:
@jahaniam thanks man ! it works.

LMWafer, 2022-01-04:
I created a docker image to run slam with Realsense cameras (D435i etc.). Here is the [Docker hub link](https://hub.docker.com/r/lmwafer/orb-slam-3-ready), it's empty for the moment but I'm working on a robust documentation.

richtong, 2022-06-21:
@LMWafer that is great, I updated the @jahaniam to work properly on an M1 mac and have shipped a PR, but I am not sure how to get the Dockerfile updated